kennethjiang/OctoPrint-Slicer #1: Not able to close X/Y/Z rotation bubble

derpicknicker1, 2016-09-26:
When i hit the rotate button on the left a bubble with x/y/z input. But i'm not able to close this.
* I am using Chrome Version 53.0.2785.116 m (64-bit)
* Octoprint version is 1.2.16 (master branch)

kennethjiang, 2016-09-26:
It's on my list and I will get to it at some point.

The top 2 features on my list are:
- scale the model
- advanced settings such as "enable retraction".

Would you rank closing rotation bubble higher or lower than these 2 features above? If you'd rank it higher I will probably address it first.

derpicknicker1, 2016-10-01:
Do whatever you like. I think the plugin is fine in regards to its phase of development.
Ideas:
* A function to move the object arround would be nice.
* Also a visivble mark on the build platform to show where "the front" of the printer is.

kennethjiang, 2016-10-28:
@derpicknicker1 now you can toggle the bubble just by clicking the button again. Also I made enhancements that address your other suggestions. Happy slicing. :)

commit c264f37f (2016-10-28): fixes #1. Toggle value inputs on click
closed 2016-10-28 by kennethjiang