kennethjiang/OctoPrint-Slicer #5: Move the object

universam1, 2016-10-02:
Awesome work, thanks for providing this great addition!

Maybe I'm missing something, but the abitlity to move the object is super critical IMHO.
Is there any support for it?

Thank you!

kennethjiang, 2016-10-03:
cool. I think "moving mode" is now higher on the list. :) However, I tried to implement it before and it turned out to be quite tricky to do. So it'll take a while. Stay tuned.

universam1, 2016-10-04:
The problem happens especially with bigger objects that are not centered or the origin is at 0,0 so no way to print within the boundaries...

Great, looking forward! :smile:

donovan6000, 2016-10-14:
It shouldn't be too tricky to implement. All the slicer plugins for OctoPrint will [center whatever it slices](https://github.com/ra100/OctoPrint-Slic3r/blob/master/octoprint_slic3r/__init__.py#L201-L202) in [the middle of the bed](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/cura/profile.py#L820-L836), so you just need to adjust the bed's width and depth in the printer profile before slicing so that the sliced objects gets placed where you want it to be, then restore the original bed's width and depth after it finishes. Actually your plugin [already contains M33 Fio's code verbatim that does this](https://github.com/kennethjiang/OctoPrint-Slicer/blob/master/octoprint_slicer/__init__.py#L281-L283), but your not using [OctoPrint's BlueprintPlugin mixin](http://docs.octoprint.org/en/master/plugins/mixins.html#blueprintplugin) to utilize it.

Also why did you dumb down [M33 Fio's](https://github.com/donovan6000/M33-Fio) model editor so much?? The original has multi model support, moving, rotation, scaling, merging, cutting, cloning, etc, and you stripped it down to only rotating and scaling.
[image]

closed 2016-10-14 by kennethjiang

kennethjiang, 2016-10-14:
Actually I was able to implement it with a much simpler design than M33, thanks to the built-in 'position' parameter that Octoprint supports.

Thank you for telling me about OctoPrint's BlueprintPlugin mixin. I will take a look at it to see how much I can leverage to make my code simpler.