kennethjiang/OctoPrint-Slicer #6: [????] Unknown bug?

bolsoncerrado, 2016-10-03:
i first blamed TouchUI but we proved wrong and upon uninstallation of the Slicer, the system worked fine again.

This is what I got after installing Slicer and restarting the service:

[image]

Apparently it fails only locally, not remotely.

It tries to load WebGL and that is not supported on the RaspberryPi Chromium browser. Hence the function getExtension is not supported, it seems the library Three.js is throwing this error.

Basically there is no check if the browser supports WebGL, what breaks all other features of OctoPrint and TouchUI.

kennethjiang, 2016-10-03:
Ah I never tested it on Chromium with touch UI. Will test it and fix the problem. I assume Slicer plugin should just disable itself at the absence of WebGL. Let me know if you have better ideas.

bolsoncerrado, 2016-10-06:
If its a limitation of the browser itself, then yeah, few things u can do about it...

bolsoncerrado, 2016-10-06:
....or....

http://raspberrypi.stackexchange.com/questions/49690/chromium-50-webgl-support-on-jessie

hummm worth trying?

kennethjiang, 2016-10-07:
Interesting. But I think it's a bit too intrusive for my plugin to change Chromium settings in /etc.

I guess I will go with the fix of disable my plugin if webgl is not present.

bolsoncerrado, 2016-10-07:
May be a switch to give the  user a chance to choose? :D

kennethjiang, 2016-10-07:
If users want, they can always make the changes themselves. I don't feel it's the right decision to change another app's settings "for the users".

I just pushed a new version to disable slicer when webGL is not supported. If you want to use the Slicer Plugin in Chromium, you can follow the instructions you found to enable WebGL.

Let me know if it's still breaking your OctoPrint/TouchUI. If not, I will close this issue.

closed 2016-10-10 by kennethjiang