kennethjiang/OctoPrint-Slicer #9: Slice multiple models at a time

kennethjiang, 2016-10-11:


universam1, 2016-12-23:
That would be awesome!
Thanks for great work with this project!

eyal0, 2017-01-24:
I'm working on it.  There is a somewhat working version here:

https://github.com/eyal0/OctoPrint-Slicer/archive/multiple_stl.zip

I'm working on it so it might not be stable.

* Click on the magic wand to add an stl file to the platform.
* Click on a model on the platform to select it for moving, rotating, etc.
* No way to delete models yet but I'm working on it.  For now, just reload (F5) in the browser.
* I hope to add auto arrange and detection of overlapping models, too.

kennethjiang, 2017-01-25:
again my apology for the delay due to my schedule. can't wait to see it in action!

kennethjiang, 2017-02-01:
I tried it and it looks good! Can you open a PR so that i can take a look at the diff?

eyal0, 2017-02-02:
Sure, but there are multiple steps.  First, OctoPrint-Slicer needs to support uploading temp STLs.

Currently, if you scale or translate a model on the platform, when you go to print, OctoPrint-Slicer will upload the modified model, *overwriting* the model already on the server.  The deletes the original STL.

If I left this behavior, it would cause the multi-model platform to overwrite the STL with the single model.  Not a good solution.

So instead, I made it so that modified STL files are uploaded into a temporary file and then sliced from there.  When slicing is complete, the temporary file is deleted.  I think that this is a good solution.

After that is set, then it's possible to put multiple STL on the platform.  I'll send the PR for that.

kennethjiang, 2017-02-03:
Big thanks to @eyal0 for getting this challenging issue nailed!

closed 2017-02-03 by kennethjiang