kennethjiang/OctoPrint-Slicer #12: Print not centered

jangrewe, 2016-10-17:
Hi, thanks for your plugin, i've been longing for something like this for a very long time!
It works very well, except that it doesn't center the generated model on the bed. That's fine if the print isn't that big, but now i have something that needs 225mm width (i have a Prusa i3 MK2 with 250mm) and the generated gcode places the model partially off of the left edge of the bed.
I have slice the very same STL with Cura directly on my desktop, and with the same profile it generates gcode that puts the model right in the center of the bed.

jangrewe, 2016-10-17:
Nevermind, i just saw that you released 0.7.1 which fixes this. You forgot to update `setup.py` with the version number, though ;-)

closed 2016-10-17 by jangrewe

kennethjiang, 2016-10-17:
Awesome. Great minds think alike. :)

kennethjiang, 2016-10-17:
updated setup.py. thank you for pointing it out!