kennethjiang/OctoPrint-Slicer #13: JS Error: Unable to process binding "options: function (){return matchingSlicers }

LVNCELOT, 2016-10-26:
After installing the slicer and rebooting the system, the JS code fails with this error, rendering me unable to login to the web app and uninstall the plugin.

kennethjiang, 2016-10-26:
What browser/version are you using? Also what OctoPrint version are you using (this may be difficult to get if you can't get into OctoPrint).

A few suggestions to get around this problem:

1. Try a different browser. I have been testing OctoPrint and this plugin on Chrome 53.0.
2. (if you are comfortable with command line) SSH to your Raspberry Pi (ssh pi@octopi.local), run command "/home/pi/oprint/bin/pip uninstall --yes OctoPrint-Slicer"

kennethjiang, 2016-10-28:
@LVNCELOT Can you please provide the info on what browser and which version you were using when you had that error. I tested my plug in on latest version of Chrome, Firefox and IE11. I'll be happy to fix it if you can offer more info about this issue.

derpicknicker1, 2016-10-31:
Maybe you should also test in Safari to make Mac users happy :)

LVNCELOT, 2016-10-31:
@kennethjiang Please excuse the late reply. The issue arose on 1.2.11 on both Chrome and Firefox. I've since reinstalled octoprint and on 1.2.16 everything is working fine.

kennethjiang, 2016-10-31:
Thank you so much @LVNCELOT for pointing out this compatibility issue. Based on your info I found out that Slicer plugin requires OctoPrint 1.2.14 and up. I have created a PR (https://github.com/OctoPrint/plugins.octoprint.org/pull/47) and hopefully it'll get merged soon.

Once my PR get merged I will close this issue. But let me know if there is anything else.

closed 2016-10-31 by kennethjiang