kennethjiang/OctoPrint-Slicer #16: No updates via Update Manager

jangrewe, 2016-11-11:
I noticed that your plugin doesn't have any reference back to this repo in OctoPrint's Update Manager, so you don't get any updates unless you install them manually. I'm guessing that's fixed easily with some additional info in setup.py, but i can't offer a solution right away ;-)

derpicknicker1, 2016-11-18:
this is because the release names start with 'v'. Had the same problem with telegram plugin some time ago.

kennethjiang, 2016-12-01:
Thanks folks for initiating/commenting on this issue. And sorry for the delay as I was just back from a long vacation.

@derpicknicker1 I tested release names (tags) without the starting 'v' and somehow they still didn't show up in OctoPrint's "Settings" -> "Software Update". Please see attached screenshot. Will appreciate it if you can shed some light on what I'm still missing.

[image]

kennethjiang, 2016-12-03:
@jangrewe I just tested it and it seems to work - thanks to the tip from @derpicknicker1 . However I'm not 100% confident so I'm not closing this issue now. Can you test it to see if it works for you and let me know?

jangrewe, 2016-12-03:
@kennethjiang all my OctoPrint instances updated themselves to 0.8.1, looks like you fixed it! =)

closed 2016-12-03 by jangrewe

kennethjiang, 2016-12-23:
Thank you @derpicknicker1 . I dropped 'v' from my tag and started using pure numeric version numbers a while ago. However when I pushed a new version it almost never showed up immediately in my other OctoPrint installations. How long do you have to wait for the new versions to show up?

derpicknicker1, 2016-12-23:
This is because of Octoprint is caching the update informations. 
go to ``Settings -> Software Updates`` select ``Advanced Options`` at the bottom of the dialog and force the refresh of update data.
I don't know how often Octoprint refreshes the update data. Maybe every 24h?

kennethjiang, 2016-12-24:
cool I will try "force refresh" when I have next version. Thanks @derpicknicker1