kennethjiang/OctoPrint-Slicer #18: Using a round delta printer with origin set to 'center' in the printer profile cause off set issues

JorgMU, 2016-11-19:
If I slice with a printer profile that has the origin set to 'center', and a printer area of 200x200x300 the gcode generated puts the part completely off the bed to the top-right when viewed in the gcode viewer.

Using the built in slicer without any other plugins works perfectly.

kennethjiang, 2016-12-01:
Sorry for the delay. Just came back from a long vacation.

Will work on this issue whenever I get a chance.

JorgMU, 2016-12-01:
No worries, thank you for looking into it!

Thanks,

Jorg

Erik 'Jorg' Jorgensen
Support Systems Administrator - Principal
Vended Applications Support Services
Division of IT
University of Missouri - Columbia
jorgie@missouri.edu<mailto:jorgie@missouri.edu>
573.882.5974
"If you're not gonna tell the truth, then why start talking?" - Gene Wilder, 1933 - 2016 R.I.P.

commit 5ad74452 (2016-12-02): closes #18. print bed offset should be (0,0) when origin is at the center
closed 2016-12-02 by kennethjiang