kennethjiang/OctoPrint-Slicer #19: Skirt

analogmonster, 2016-12-10:
Hi, can you please add an option for skirt as well as brim? I use skirt when it's a model that won't have adhesion problems l, just to prime the nozzle and brim when it's needed. Thanks

kennethjiang, 2016-12-11:
I think when you choose "none" for adhesion, Cura engine will generate "skirt" for you. Can you try it to see if it's what you want?

analogmonster, 2016-12-11:
Yeah you are right. Sorry this is my second octoprint install, but previously was using an m3d micro so was using the m33fio plugin for slicing and that had a tick box for skirt brim or raft. Thanks

closed 2016-12-11 by analogmonster

donovan6000, 2016-12-11:
@analogmonster In case you didn't know, you can use M33 Fio with all types of printers. Just enable the Settings > M33 Fio > "Not using a Micro 3D printer" settings, and then all the Micro 3D specific features will be disabled. With that settings enabled the slicer profiled editor lets you set additional settings like the print temperature and bed temperature, and the model editor displays a grid proportional to the size defined in the printer's profile instead of the Micro 3D printer frame.
[image]

analogmonster, 2016-12-11:
Hmm thanks I may well do that, still have the m3d but it's relegated to short prints until I build a Bowden system I'm happy with. But kind of makes sense to have the same system across my printers, thanks