kennethjiang/OctoPrint-Slicer #22: Brakes Octoprint

Ytgr, 2016-12-30:
First of all I'd just like to say thank you for developing anything for people to have and change on their own for free. I'll just cut right to the chase - I have a brand new install of OctoPi and whenever I install this plugin via the url address in the plugin manager, I restart OctoPrint, and then it is broken. I can't login or anything. If you can help me or if you need more information please let me know. Thanks in advance.

derpicknicker1, 2016-12-30:
I had the same issue. For me it was a problem with WebGL. My browser (Chrome) didn't support it. After activating some settings and flags in Chrome, it worked for. [THIS](http://superuser.com/questions/836832/how-can-i-enable-webgl-in-my-browser) helped me.
To fix your broken octoprint, ssh into your rPi and open config.yaml for edit. Search for the 'plugins' section in the file and insert this:
```
plugins:
  _disabled:
  - slicer
```
Note: There are two spaces before ``_disabled`` and ``- slicer``

One question at the end: Why do you install via URL and not via plugin repository?

After this change, restart octoprint.

Ytgr, 2016-12-30:
Thanks for the help and reply. I was installing via the URL because when I searched for the plugin in the manager it wouldn't show up. Then I updated to the latest OctoPrint, the plugin showed up, I installed it, and it worked.

closed 2016-12-30 by Ytgr

kennethjiang, 2016-12-30:
Thank you so much @derpicknicker1  for fielding this issue for me!