kennethjiang/OctoPrint-Slicer #23: Slicer doesn't handle folder structure good

bizziebis, 2017-01-01:
With OctoPrint 1.3.0 a possibility to create folders inside the upload folder was introduced.

When you create a folder and and enter it, then upload a .stl, the slicer plugin gives you a wrong output path. It shows something like "Vases/Vase_1.gco". This will  result in the creation of a file called "VasesVase_1.gco" in the currently active folder.

Side effect of this problem is that the Alalysis is not performed afterwards. I think this is related to the wrong creation path/filename.

This will result in a gcode file which has no proper metadata and is missing filament/volume properties. Plugins which depend on this information like the History plugin and the Costs plugin will fail.

bizziebis, 2017-01-01:
Darn, directly after opening this issue I see that my plugin is still at 0.8.1, not at 0.8.3. I will check back later when I get it updated.

bizziebis, 2017-01-01:
Solved, sorry for the confusion :)

closed 2017-01-01 by bizziebis

kennethjiang, 2017-01-01:
Glad that you figured this out!