kennethjiang/OctoPrint-Slicer #24: No file analysis after slicing

bizziebis, 2017-01-01:
After a file is sliced and saved as .gco, no analysis is performed. The metadata is missing filament/volume information afterwards.

When I download the sliced file, delete it from OctoPrint and re-upload it, the analysis is started and the correct metadata is created.

After slicing:

>  Marvin_KeyChain.gco
> Uploaded: a few seconds ago
> Size: 998.9KB
> Estimated print time: 18 minutes
> Cost: €NaN
> 

After upload:
`octoprint.filemanager.analysis - INFO - Starting analysis of local:testing/Marvin_KeyChain.gco`

>  Marvin_KeyChain.gco
> Uploaded: a few seconds ago
> Size: 998.9KB
> Model size: 112.59mm × 109.48mm × 25.00mm
> Filament: 0.83m / 2.00cm³
> Estimated print time: 18 minutes
> Cost: €0.07

OctoPrint 1.3.0
Slicer plugin 0.8.3

kennethjiang, 2017-01-03:
Do you use a plugin to show the cost? I'm also on OctoPrint 1.3.0 but I don't see things like Estimated print time or Cost.

bizziebis, 2017-01-03:
Yes I use [Cost](https://github.com/jasiek/OctoPrint-Cost) plugin to show the costs of each print. It's just injected javascript code which doesn't interfere. Also with that plugin disabled I get no filament usage information.
Estimated print time was always there if I remember correctly.

kennethjiang, 2017-01-04:
I installed Cost plugin and was able to reproduce the issue you reported.

However I suspect this is a bug of either Cost plugin or OctoPrint itself. Because if I uninstall Slicer plugin and use OctoPrint built-in slicer to slice a model, I have exactly the same issue.

Can you do the same (uninstall Slicer plugin and slice a model with OctoPrint built-in slicer) and see if you experience the same issue? If so, I suggest you to check with the author of Cost plugin.

derpicknicker1, 2017-01-04:
The cost plugin simply gets file metadata (filament usage and estimated print time) and calculates the costs. There is no way to interfere with fileanalysis. I sliced with the build in slicer option and the metadata (see below) will not include filament usage (tool0 and tool1 are empty). Same result if i use the slicer plugin.  I think this is a Octoprint issue or the analysis has to be triggerd in some way after slicing is done.

```
lego_deckel_test.gco:
  analysis:
    estimatedPrintTime: 5816
    filament:
      tool0: {}
      tool1: {}
  hash: 45698ee20bf37128dab8fbf5e499083a491852ee
  links:
  - hash: 666c9407805911c4aeb264fa57f24f98cb3d3451
    name: lego_deckel.stl
    rel: model
```

I reported this issue: https://github.com/foosel/OctoPrint/issues/1685

bizziebis, 2017-01-04:
@kennethjiang You are right. The problem is also with the build-in slicer. So It's not actually a problem of your plugin.

But as derpicknicker1 mentioned it's also not a problem of the Cost plugin. It's a OctoPrint bug I'll be watching the new issue created at the OctoPrint git for a solution.

Thank you both for checking this issue!

kennethjiang, 2017-01-04:
Thanks @derpicknicker1 for helping me around!

closed 2017-01-04 by kennethjiang

derpicknicker1, 2017-01-11:
Gina announced that this issue will be fixed in 1.3.1

kennethjiang, 2017-01-11:
Cool! Thanks again @derpicknicker1