kennethjiang/OctoPrint-Slicer #25: Slicer doesn't work in Chrome?

RiaanAspeling, 2017-01-14:
It seems slicer "freezes" when activated in Chrome (Version 55.0.2883.87 m). It woks find in Edge and Firefox.

derpicknicker1, 2017-01-14:
For me it works fine in Chrome Version 55.0.2883.87 m (64-bit).
Slicer Version 0.8.3

kennethjiang, 2017-01-14:
It doesn't sound like a Chrome version problem. Do you know how to use Chrome Developer Tool @RiaanAspeling? If so can you do some quick trouble-shooting to see where it get stuck?

RiaanAspeling, 2017-01-15:
@kennethjiang : Just basicly using F12 in Chrome. Or do I need something like Fiddler?

kennethjiang, 2017-01-16:
Hmmm, I'm out of ideas how I can debug this issue for you. Is there a way I can access your Octoprint and see what's going on? The problem is I can't reproduce what you are experiencing... :(

RiaanAspeling, 2017-01-16:
@kennethjiang : I will investigate further and let you know if I find anything, thank you. If I can't manage to find anything I'll arrange a port forward with DDNS and get you access to the Pi.

kennethjiang, 2017-01-16:
Sounds good!

kennethjiang, 2017-02-01:
Closing it for now. @RiaanAspeling if the issue persists and you find a way to reproduce it, please re-open this issue.

closed 2017-02-01 by kennethjiang