kennethjiang/OctoPrint-Slicer #27: No slicers available

sanchosk, 2017-01-15:
Hi.

I have a problem.
I have the plugin installed, but I cannot find any configuration. It does not show in my settings menu.
I can't find any way how to configure it and add some slicers :(
Can you, plz, advice?

kennethjiang, 2017-01-16:
Slicer plugin requires no settings. Every time you slice a model, either by pressing the slice button or uploading a new STL, this slicer plugin (as opposed to the simple slicer built into OctoPrint) will run automatically.

sanchosk, 2017-01-16:
And by "simple slicer" you mean Cura plugin? Why is there slover selection then?
I don't want to sound negative - I just hoped this would allow me to use slic3r on my octoprint device - which would be amazing!

kennethjiang, 2017-01-16:
Unfortunately my plugin doesn't bring in the support of slic3r.

closed 2017-01-22 by kennethjiang