kennethjiang/OctoPrint-Slicer #28: Request: Retraction/Remember Settings

SoyBlack, 2017-01-19:
Please consider allowing for:

1. Set Retraction - Speed and Distance
2. Default Settings - I use PLA a lot and use the same settings often. Would be nice to have the ability to make a PLA preset that I can tweek slightly as needed - rather than having to re-enter ALL of the settings on each print.

Otherwise - thanks for the excellent plugin!

kennethjiang, 2017-01-22:
sorry for the delay!

I will look into setting speed and distance for retraction.

You can upload slicing profiles and select the desired one from the "General" tab in "Slicer". For details on how you can upload profiles please check: https://github.com/foosel/OctoPrint/wiki/Plugin:-Cura

closed 2017-01-23 by kennethjiang

kennethjiang, 2017-01-23:
@SoyBlack retraction speed and distance have been added to version 0.8.4

SoyBlack, 2017-01-23:
Awesome - thanks! I will update once my print is done here and check it out!

commit 058fca3a (2017-01-23): closes #28.