kennethjiang/OctoPrint-Slicer #46: File Path

thjansen72, 2017-02-16:
Is it possible to change the gcode-output-path back to the old behavior?
In the older versions, the gcode file was written in the folder  of the STL - very good for sorting.
THe newer version  (>0.9) is writing it in the Files-"root" - no matter where the stl  is - and must be mover manualy for sorting.

kennethjiang, 2017-02-17:
Ah I think this is a bug that was introduced recently.

The latest version allows you to load multiple STL files into slicer and slice them at once. Can you consider this scenario: if you load **folder1/A.stl** into slicer first, then **folder2/B.stl** into slicer. There are 2 options to save gcode to:

1. **folder1/A.gco**
2. **/A.gco** (root folder)

Which option do you prefer?

thjansen72, 2017-02-17:
Damn (sorry)! Of cause you're right. It wasn't clear to me. TNX

kennethjiang, 2017-02-17:
Oh I didn't mean to say you were wrong. I will change the code to use the correct folder when there is only 1 STL file. And I'd like to hear your opinion on where you'd expect the Slicer plugin to put the gcode when there are multiple STL files. I'd like it to work in a way that is the most intuitive to most users.

thjansen72, 2017-02-17:
O.K. I think it is a question of workflow. At the moment, the name of the first stl is used and the main folder. It sounds logical to me to put the gcode in the same directory as the first STL: start to prepare printing, adding the first stl (the one I want to print primary) then I'd find out: oh there is a lot of space for more and would add the next stl as second... I think in a bigger projekt with multiple parts (stls) I , start with the biggest and adding the smaler ones- at my place they are all in same subdirectory. Mixed gcode files from different projects doesen't matter much -  I delete them after successful printing - they are just for a longer printing time (over the night).
This is just my opinion and my workflow.

kennethjiang, 2017-02-17:
Thank you so much for your inputs @thjansen72 . I'm impressed how you approached it thinking of it in terms of workflow.

I'm a software engineer and I love creating software for 3D printing. But I'm probably less experienced in 3D printing itself than people like you. Do you mind a phone call/skype with me so that I can learn from your experience to be able to incorporate better features and user experience into my software?

thjansen72, 2017-02-18:
I think, what fmodesto wrote is a good beginning for a possible workflow. The single file in the "old way" and with adding the next file the question: "what do you wan't" and "what's the ne name to put it to".
O.K. Skype is running most time thomas@thjansen.de

closed 2017-02-19 by kennethjiang

kennethjiang, 2017-02-19:
@thjansen72 I fixed this bug. Please wait for next release and verify if it works as you expected.

I'm still struggling to figure out the best work flow for multiple STLs. What @fmodesto suggested seems to make sense. But the only problem is that **arrange** is a button. If I make it a toggle, I don't know how users would find a way to manually trigger auto-arrange function.

Sent a request to your skype ID. Will really appreciate it if we can connect on Skype so that I can get your detailed feedback and suggestions.

commit b384cff0 (2017-02-19): closes #46. refactor how file names are handled