kennethjiang/OctoPrint-Slicer #49: Control addition of multiple objects

fmodesto, 2017-02-18:
With the last release is it possible to slice multiple objects and print them in one go. It is a great feature but should not be enabled by default, or should be a way to control de behaviour. I have already screw some prints thanks to the left overs.

I think it would be good if the arrange button can be toggled and in that mode append the files, otherwise ignore the previous content and start from zero.

Proposed normal workflow:
1- Upload STL file
2- Slice it
3- Print
4- Upload STL file (previous content are discarded by default)
5- Slice it
6- Print

Slice multiple objects workflow:
1- Upload STL file
2- Toggle arrange button
3- Upload STL file (should append to the existing slice)
4- Repeat...

kennethjiang, 2017-02-18:
Hmmm, I think current behavior is similar to how Cura works - it requires users to explicitly clear all models, otherwise new model will be added to existing ones.

Can you give me more details on how current flow confused you?  I'd like to figure out an optimal user experience for most users.

BTW **arrange** button is only optional to use when there are multiple models on the bed. You can always manually move models around without resorting to **arrange** button.

kennethjiang, 2017-02-20:
@fmodesto I just released 0.9.4 in an attempt to make the flow less confusing. Can you please upgrade it and let me know if the new flow improves the situation? If so, please close this issue. Otherwise feel free to make further suggestions.

fmodesto, 2017-02-20:
Works great, thank you :)

closed 2017-02-20 by fmodesto