kennethjiang/OctoPrint-Slicer #51: G code is not compatible with printer

hieronymousch, 2017-02-20:
Used a working config bundle from slic3r 1.2.9, imported without problem.
Had to modify the temperature & G-codes (which shouldn't have to be done as this information is in the config_bundle
G code is half the size of the slic3r produced code on my desktop.
Printer won't print

kennethjiang, 2017-02-20:
Thank you for submitting an issue.

Can you provide me with the following?

1. the STL file
2. the generated G-code 
3. The config bundle

I'll try to reproduce the problem.

hieronymousch, 2017-02-20:
Hi,

Thanks for your quick reply.

Files in attachment. I already cleaned the config bundle but same results.

Renaming the printer profile name didn't solve the problem neither.

Grtz,

J

# generated by Slic3r 1.2.9 on Mon Feb 20 20:07:36 2017

[filament:20170120-white-ice]
bed_temperature = 0
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 3
extrusion_multiplier = 0.9
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_diameter = 1.75
first_layer_bed_temperature = 0
first_layer_temperature = 205
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
slowdown_below_layer_time = 5
temperature = 205

[presets]
filament = 20170120-white-ice.ini
print = 20160418.ini
printer = K8400.ini

[print:20160418]
avoid_crossing_perimeters = 0
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 3
complete_objects = 0
default_acceleration = 0
dont_support_bridges = 1
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0
fill_angle = 45
fill_density = 20%
fill_pattern = 3dhoneycomb
first_layer_acceleration = 0
first_layer_extrusion_width = 200%
first_layer_height = 0.35
first_layer_speed = 15
gap_fill_speed = 20
gcode_comments = 0
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
[... 2027 lines cut ...]
G1 X158.590 Y146.822 E43.46032
G1 X163.767 Y142.848 E43.73283
G1 X164.281 Y142.593 F7800.000
G1 X167.610 Y144.901 F7800.000
G1 X165.202 Y144.256 E43.91595 F3600.000
G1 X162.335 Y146.456 E44.18151
G1 X163.355 Y150.266 E44.47124
G1 X165.983 Y150.970 E44.67113
G1 X166.044 Y151.195 E44.68824
G1 X166.857 Y151.194 E44.74797
G1 X166.742 Y150.767 E44.78051
G1 X168.666 Y148.843 E44.98040
G1 X167.610 Y144.901 E45.28023
G1 X170.178 Y138.445 F7800.000
G1 X165.718 Y134.019 E45.74186 F3600.000
G1 X165.552 Y134.185 E45.75912
G1 X166.608 Y138.127 E46.05895
G1 X169.236 Y138.832 E46.25884
G1 X170.292 Y142.774 E46.55867
G1 X168.368 Y144.697 E46.75856
G1 X169.424 Y148.639 E47.05839
G1 X171.253 Y149.129 E47.19746
G1 X171.287 Y142.633 E47.67471
G1 X171.051 Y142.570 E47.69264
G1 X1

kennethjiang, 2017-02-21:
I attached the gcode generated on my OctoPrint using Slic3r as the engine. It almost doubles the size of the file you sent me. Can you verify if this gcode is printable?

In the gcode you sent me, I see:
```
    5 ; perimeters extrusion width = 0.84mm
    6 ; infill extrusion width = 0.84mm
    7 ; solid infill extrusion width = 0.84mm
    8 ; top infill extrusion width = 0.84mm
```

However, when I sliced it using the config init file you attached, the resulted gcode is:
```
    5 ; perimeters extrusion width = 0.58mm
    6 ; infill extrusion width = 0.58mm
    7 ; solid infill extrusion width = 0.58mm
    8 ; top infill extrusion width = 0.58mm
```
Do you know why yours seems to be sliced using different width settings?

[3b2072ef0f7d7441e2d1fbc319fdbce6-Coat-hook.gco.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/791674/3b2072ef0f7d7441e2d1fbc319fdbce6-Coat-hook.gco.zip)

hieronymousch, 2017-02-22:
Hi,

Your gcode prints fine (is printing now) so the problem seems to be on my side.
I just imported the slicing profile (from Slic3r 1.2.9) and sliced via octoprint. No idea why the extrusion widths were changed.

kennethjiang, 2017-02-22:
Did it go well when you imported the profile and sliced it again via octoprint? Let me know if it didn't.

kennethjiang, 2017-02-24:
Closing it on the assumption that user no longer experience problems.

closed 2017-02-24 by kennethjiang

hieronymousch, 2017-02-24:
Hi. 
Import worked fine.. but after some more digging I noticed that I've installed the dev version of the slicr, I presume that the errors are caused by that (having the same problem with another RPI) with the normal slic3r gui. 
Thanks for your help and my apologies for bothering you with this noob mistake ;)

kennethjiang, 2017-02-24:
No apologies needed. By working with you on this issue I have learned something too!

Feel free to re-open this issue or file another one should you keep having issues.