kennethjiang/OctoPrint-Slicer #52: JS error in Chrome

eliasbakken, 2017-02-21:
There seems to be a javascript error occuring in Chrome but not FireFox. 

Chrome reports this line as faulty: 
https://github.com/kennethjiang/OctoPrint-Slicer/blob/133fc7ff9bd0cb3f59bfe44f72ae1fe389152488/octoprint_slicer/static/js/RectanglePacker.js#L22
[image]

kennethjiang, 2017-02-21:
@eliasbakken I will look into what was causing the issue. But first of all I'm very interested in learning how you got Chrome to tell you which file/which line caused an error? I have been searching for such a solution but haven't found a good one, since all individual JS files got concatenated into 1 big JS file.

Can you give me a bit more info on the steps leading to this error? I just tried it in Chrome 56.0.2924.87 and Firefox 51.0.1 and didn't see any error.

kennethjiang, 2017-02-23:
@eliasbakken any more info on this issue? My Chrome (56.0.2924.87 on OS X) is fine with this line https://github.com/kennethjiang/OctoPrint-Slicer/blob/133fc7ff9bd0cb3f59bfe44f72ae1fe389152488/octoprint_slicer/static/js/RectanglePacker.js#L22

kennethjiang, 2017-02-24:
Hi @eliasbakken would you kindly provide details on how to reproduce this error, and how you determined it was caused by that line of code you specified?

eliasbakken, 2017-02-26:
Sure! I'm on Chrome  48.0.2564.109, which is the standard for Ubuntu 16.10

[image]

Here is a more compete screenshot. I use he built in debugger in Chrome (Ctrl+Shift+i) which tells me there is an error in packed_plugins.js line 7900. For some reason I have to scroll down to the relevant line, it does scroll to the right line by itself.

kennethjiang, 2017-02-27:
Ok I will try it with Chrome on Ubuntu.

kennethjiang, 2017-02-27:
I tried it on Chrome 56.0.2924/ubuntu 16.04 and couldn't reproduce this error.

Downloaded Chrome from https://www.google.com/chrome/browser/desktop/. If you don't mind upgrading chrome you can do it to see if error goes away.  When i get a chance i will find a way to install older version of Chrome to see if I can reproduce the issue.

eliasbakken, 2017-02-27:
Upgraded to 56.0.2924.87, not a problem any more. 

It's up to you how severe you think this bug is. Maybe write a notice about it somewhere or get someone to fix it. I'd love to help out, but I really don't have the time right now. Love this plugin and want to see it succeed:) We are adding it as default to the Kamikaze image, starting from 2.1: http://wiki.thing-printer.com/index.php?title=Kamikaze

kennethjiang, 2017-03-04:
Found out what the problem was. It will be fixed with next version release.

closed 2017-03-07 by kennethjiang