kennethjiang/OctoPrint-Slicer #53: JQuery - no knockout.js

Jabarnes01, 2017-02-22:
Would it be possible to port the knockout.js to plain old JQuery?

kennethjiang, 2017-02-23:
OctoPrint chose knockout therefore any plugin will have to work with knockout to be available.

Is knockout causing any problem?

Jabarnes01, 2017-02-24:
No but I want to bring this out of octoprint. I want to replace the input and output endpoints with custom endpoints.

kennethjiang, 2017-02-24:
Ah I see your point now!

Actually I have been thinking in the similar direction. I'd love to package some reusable components, event independent of jQuery (ideally they shouldn't be attached to specific html tags so jQuery should be out of the picture too).

If you want to take a stab at it I will love to collaborate on the effort.

kennethjiang, 2017-02-25:
Actually most of the JS files in folder `octoPrint_Slicer/static/js` are not tied to knockout. Only `slicer.js` and `profile_overrides.js` do.

`STLViewPort.js` is the critical file that renders STL models in 3js. It doesn't even depend on jQuery. You can start using it now if you don't care it isn't nicely packaged. Let me know if you need my assistance on that.

Jabarnes01, 2017-02-27:
Ok, I'm going to take a stab at this... I'm going to rework slicer.js and profile_overrides.js

Jabarnes01, 2017-02-28:
I don't see basic or any advanced options after I installed the plugin... Wondering why that is?

kennethjiang, 2017-02-28:
The options are not shown until you load a model. It's probably neither necessary nor intuitive. Will open an issue for that.

Jabarnes01, 2017-03-01:
I had a model loaded but the options did not show up. No big deal... My other question is why are you using stlViewPort.js? I'm recreating everything from scratch and so far it's not necessary to have this lib. Is the library depreciated?

Jabarnes01, 2017-03-01:
I see you created that has a helper for custom functionality. Got it. I'm having a problem when I use the transform inputs the model disappears on my custom app. Not sure why that happens yet...

kennethjiang, 2017-03-14:
@Jabarnes01 In case you will find it useful, I am in the process of separating some common 3D model manipulation into [an independent package](https://github.com/kennethjiang/3tk). The bulk of what the slicer plugin currently relies on will move to 3tk, which depends on neither knockout nor jQuery.

I will close this issue unless you have anything that's beyond the scope of what I'm doing with 3tk.

closed 2017-03-14 by kennethjiang