kennethjiang/OctoPrint-Slicer #62: Optimize user interaction

kennethjiang, 2017-03-07:
- zoom on mouse scroll is too sensitive
- orbiting is a bit sluggish: why is this example so responsive? https://stemkoski.github.io/Three.js/Model-Animation.html

scruss, 2017-03-14:
Could the numeric fields be assigned a tab order so you could update one, hit TAB, and go to the next field down please?

kennethjiang, 2017-03-15:
@scruss Good suggestion! 

And you are more than welcome to a stab at it and I will merge the PR. I've been quite swamped lately so it'll take me a while to get around to it.

John-Mc, 2017-03-15:
I like the "Split" button. Is it new? I didn't notice it before.

A couple of user interface suggestions:

It makes more sense to me to keep the "Remove" button at the bottom of the column of buttons. It's used more frequently than Split. It also just seems to "make more sense" to me on the bottom. But that could be just me.

Using the "Split" button on a model just to see what it does, I ended up with about 20 small parts scattered across the bed. I had to select each of these individually and remove them. A way to clear the whole bed would be nice, or perhaps a way to "Select All" (and/or select more than one object but not necessarily all) and then clear would be helpful.

kennethjiang, 2017-03-16:
Appreciated your feedback @John-Mc . The "split" button is supposed to break down models that consist of unconnected parts. But because of [this bug](https://github.com/kennethjiang/OctoPrint-Slicer/issues/67) it blows out internal structures as well. 

Would you like to delete models using *delete* key? That'll be trivial to implement. I do like the "clear all" function but I feel it may be overkill to add a button just for that. And I hesitate to add context menu (right click) to my plugin. I'm guess I'm struggling between maintaining a simple and elegant UI vs. features. I don't like how it currently prompt you every time you add an object to print bed that is not empty.

If you have some thoughts on how these user interactions can be streamlined please shoot them my way!

John-Mc, 2017-03-16:
I agree, I would not add a dedicated button: too much clutter.  What about shift-click to select more than one, or click and drag, using a "selection box" to select multiple items. That would allow more than one item to be selected without adding a button (both shift-click and the selection box are common techniques). Down side to selecting multiple then clicking remove is that people may try to do other things than remove with a multiple selection, and you may not be ready to allow that sort of thing with other operations (move, scale, etc.).

I sort of like the idea of shift or control & click on remove to remove all (or a contextual menu that offers "clear all", but I defer to your experience in this matter - I've not done a lot with designing user interfaces).

**I consider a "Clear all" a low priority**. It's not that big a deal to clear items one at a time. The bug is what made me suggest it: I wanted to see what "split" did so I clicked on it. The current part had about 20 internal structures, some rather small. It was a pain to remove them all one at a time, but once the bug is fixed so those 20 parts don't get blown out it shouldn't be an issue. (The bug could be rather confusing if one of our 5/6 students runs in to it.)

kennethjiang, 2017-03-16:
Shift-click is easily doable and pretty elegant user interaction design. "Selection box" is very common in 2D design but not so much in 3D, probably because it's hard to implement correctly. Again I appreciate all these good suggestions! @John-Mc 

I agree that the bug in *split* is confusing enough to cause a lot of trouble. I just pushed a new release that removed that button. Please upgrade when you get a chance.

closed 2017-03-25 by kennethjiang