kennethjiang/OctoPrint-Slicer #63: Supports not working (or user error)?

John-Mc, 2017-03-10:
I imported some profiles from Cura which had support turned off when importing. Loaded an STL file (See attached) on the bed which needed supports, so I turned on "everywhere" under that option, but file is printing with no supports. Did I do something wrong, or is there something else going on? Shouldn't the support setting in your slicer override the setting in the imported Cura filament profiles?
[ball in cube.stl.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/834564/ball.in.cube.stl.zip)

I apologize if this is not place to ask these sort of questions. I'm new to this, trying to help out a 5/6 grade classroom with their 3D printer. If I should be asking somewhere else, please let me know.

kennethjiang, 2017-03-10:
Thank you for submitting this issue!

I just tried to reproduce this problem myself: I saved a cura profile with Support: None. Then I turned Support: Everywhere in Slicer plugin and sliced your STL. The slicer did generate support in gcode (attached). The support is tiny (0.4mm around the bottom of the ball) though but I assume this is how Cura's algorithm works.

If you are still not getting support, please let me know and I will dig deeper into it!

[ball_in_cube.gco.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/834679/ball_in_cube.gco.zip)

John-Mc, 2017-03-10:
Thanks. There was a tiny bit of support under the ball. I assumed it would print supports under the top edges of the cube, but it did not.

I guess we can consider this one operator malfunction - I tried a simple file that I designed to need more noticeable supports, both touching build plate and within the print. Supports worked as they should. Sorry to waste your time. 

Is there a user group for this plugin where I can ask Noob questions without taking up your time?

kennethjiang, 2017-03-10:
No worries. You are always welcome to shoot questions my way by opening an issue here. I will set up a user's forum when I'm overwhelmed. :)

As a matter of fact I'll love to hear feedback from you to see how I can bring more value to you by adding features to this plugin or by creating other plugins. Are you open to a call/skype in the next few days?

John-Mc, 2017-03-10:
I'm rather new at this, so not sure how much useful input I would have. We are just getting started using OctoPi, and still having some setup issues (most related to a lack of cooperation from the School's IT folks). Is there a way to get in touch without posting a phone number or Skype address in the open?

John-Mc, 2017-03-10:
Kenneth -
    
    One thing that I believe is already on your list: a "Lay Flat"
    command. I volunteer with a 5th & 6th grade class at my local
    elementary school, trying to help them learn to use their Lulzbot
    Mini printer and offering input as they work through their designs
    (I'm no expert in that either).  The lay flat command would be
    extremely helpful. TinkerCAD is about the only 3D design program
    that will run on their ancient computers as well as on their
    ChromeBooks. It does not offer this feature. Not only is there no
    lay flat command, if they rotate their project (either accidentally
    or on purpose) there is not a good way to tell if they have it back
    flat again - zooming in and eyeballing it has it's limits. If we had
    "Lay Flat" along with your excellent slicer plugin, we could go
    right from TinkerCAD to the Pi, without having to clean things up in
    Cura on one of their laptops.
    
    Other things I'm looking for:  
    
    How to set up OctoPi to run as a WiFi Hotspot or Access Point on a
    Raspberry Pi 3, rather than logging it in to the school's network.
    The school's IT guy has been very uncooperative as we try to get
    OctoPi up and running: he won't do a static IP, he won't even give
    us the password to one of the school's WiFi networks (we're only on
    because a student happened to have picked it up somewhere). If we
    had it set up as it's own Hotspot, we could bypass the school's
    network altogether (Kids would design on TinkerCad, export to their
    classroom drive, then log on to the Pi's network and send the file
    to OctoPrint). I've found descriptions of how to do this with the
    standard Raspian installation, but that is different than the OctoPi
    set-up. Some of the files those directions say to edit don't seem to
    exist on the OctoPi image we installed on our SD card. I'm sure
    directions for this already exist somewhere, but I can't find them.
    
    John McNerney
    
    On 3/10/17 1:17 PM, Kenneth Jiang
      wrote:
    
    
      No worries. You are always welcome to shoot questions my way by
        opening an issue here. I will set up a user's forum when I'm
        overwhelmed. :)
      As a matter of fact I'll love to hear feedback from you to see
        how I can bring more value to you by adding features to this
        plugin or by creating other plugins. Are you open to a
        call/skype in the next few days?
      —
          it on GitHub, or mute
          the thread.

kennethjiang, 2017-03-10:
Really appreciated your detailed feedback John! Yes *lay flat* is on my road map and I can't wait to wrap up what I'm working on (split into parts) and get to that one. :)

I have been thinking about another plugin that may or may not address the 2nd problem you listed. Let's coordinate a time to call/skype by email: kenneth@rhumbix.com

closed 2017-03-13 by kennethjiang