kennethjiang/OctoPrint-Slicer #65: Finer Fan control

dani203, 2017-03-14:
My Printer allows for fine fan speed control, but sadly the Slicer plugin does not support that. I would like to see a slider that allows me to set the speed between 0 and 255. i run my fan at 25% most of the time, because any faster and it cools my bed down and i get terrible warping and lifting. 
Currently i need to manually download the model, edit it in notepad++ delete it from the server and reupload, which is really inconvinient.

John-Mc, 2017-03-14:
I wonder about a "tweak at Z" plugin like Cura Desktop has. Could it be used to tweak the fan speed at different heights? (I'm not requesting this. It's not a feature I've used on Cura Desktop, though others may find it useful. Just suggesting it as an option for fan control.)

kennethjiang, 2017-03-14:
@dani203 are you looking for something that Cura supports (as in the screenshot attached)? Or you need something more sophisticated as suggested by @John-Mc ?
[image]

kennethjiang, 2017-03-16:
@dani203 Please indicate if the fan settings in Cura (the screenshot that I posted) are good enough for you. If so it's quite easy to implement. Otherwise it will be very difficult.

kennethjiang, 2017-03-23:
@dani203 I'm closing this issue since I haven't heard from you. If you still want this feature, you can reopen this issue and provide more details.

dani203, 2017-04-22:
@kennethjiang Sorry, i didn't notice your reply for some reason. I don't need any sophisticated controls like increasing fan speed, all i want is to set a speed througout the entire print.
Could look something like this:
[image]

John-Mc, 2017-04-22:
@kennethjiang - if you do get into this, aren't there some filaments that benefit greatly from a different fan speed in the initial layers? If so, the image you posted on March 14 would seem to offer something for a wider range of users. 

On the other hand, @dani203's suggestion does make for a simple, uncluttered interface (unless this is done via some pop-up dialog box?)

I'm probably the wrong person to comment on this in any depth: Most of my recent printing at the school has been with HIPS, which does not seem too particular about fan speeds. When I was running other filaments, I was slicing through Cura on a laptop, and using Luzbot's default profiles, so I never adjusted this setting.

kennethjiang, 2017-04-24:
@dani203 @John-Mc sorry I was just back from vacation. And thanks a lot @dani203 for using drawing to make communication a lot easier!

I guess I will go with doing the settings that Cura currently supports. I will open another ticket to explore if it's feasible to incorporate Cura plugins which will provide much better flexibility.

kennethjiang, 2017-04-26:
@dani203 It is implemented in 1.2.5.

closed 2017-04-26 by kennethjiang

dani203, 2017-04-27:
@kennethjiang Thats great to hear, thanks for the quick implementation!