kennethjiang/OctoPrint-Slicer #68: version 1.0.2 slicer returns code 11

VictorBucha, 2017-03-19:
Experienced after update. restarting OP has no impact.

kennethjiang, 2017-03-19:
@VictorBucha Can you provide a bit more info? What part of the code returns `11`? Did you have the problem when OP was booting up, or when you tried to load an STL into the slicer? A screenshot with the error code will be great

kennethjiang, 2017-03-19:
@VictorBucha I ran into similar issue and now I understand what you mean by *code 11*. For me, this error only happens to a very specific STL. I'm investigating why it fails. Meanwhile can you help me clarify:
- Does *code 11* happen to all of your STL files, or just some of them?
- Can you attach one of the files that failed so that I dig deeper?

John-Mc, 2017-03-19:
I don't have my printer with me right now, but I am able to start OctoPrint, upload a new file and slice it in version 1.02 with no problems. The kids at school ran off a couple of prints with no problems on Friday.  I'd be happy to load and slice a few STLs at home (without the printer attached) if it would help trouble shoot. I just need to know if this is something that needs the printer attached for the error to show up, or is it a slicing error.

kennethjiang, 2017-03-20:
Thank you @John-Mc for offering to help. I believe this only happens to some STL files. Since you have not run into similar issues you don't need to test it out for me. I'll wait until @VictorBucha to get back to me with the STL file that caused the problem.

VictorBucha, 2017-03-21:
New update has no impact on the start of OP, code 11 returns only when I hit slice button. Seems to be certain stl - specific, so i'm closing the issue for now.

closed 2017-03-21 by VictorBucha