kennethjiang/OctoPrint-Slicer #69: deselecting a bad item from the whole

fidel3d, 2017-03-24:
Hi,
Would it be possible to deselect a bad printing item from the whole so it won't ruin the whole print like in pronterface

kennethjiang, 2017-03-24:
I'm not familiar with how proterface works. 

Currently you can click on the 'X' icon in slicer plugin to delete the highlighted part. But sounds like you need something different. 

Can you elaborate on how you'd de-select a bad item?

kennethjiang, 2017-03-25:
@fidel3d I just released a new version that includes some changes to how you select items. Please upgrade and tell me if it addresses your concerns.

John-Mc, 2017-03-25:
Was @fidel3D asking about deselecting and item WHILE PRINTING, so that a part that is failing can be stopped, but other parts on the bed keep printing?

kennethjiang, 2017-04-06:
@fidel3d I just release a new version. Please upgrade your plugin in to see if your problem is solved. If not, please let me know here.

kennethjiang, 2017-04-09:
I assume this issue no longer exists. You can re-open it if you still find it a problem @fidel3d

closed 2017-04-09 by kennethjiang