kennethjiang/OctoPrint-Slicer #70: Slicer window is VERY SLOW in Safari

kennethjiang, 2017-03-25:


John-Mc, 2017-03-26:
Not seeing this in the couple that I tested. For example:
  • Sliced a 2.3MB STL on Safari in 9.76 seconds.
  • Sliced same file on Firefox in 9.68 seconds

Both were running on the same computer (Mac OS 10.12.3, Firefox 52.0.1 64 bit, Safari 10.0.3)

kennethjiang, 2017-03-26:
What I mean by safari being slow is the slicer window is less responsive when you try to drag model across the print bed. 
2MB STL is probably not bad. A 10MB STL like [this](http://www.thingiverse.com/thing:763622) will make it really sluggish.
But your comments prompted me to do some more testing and found out that Safari is super slow *only* when it's in 'developer' mode (develop window is open at the bottom). In 'normal' mode it's only slightly more sluggish than Chrome or Firefox. So it's probably not as high-priority as I thought.

closed 2017-04-06 by kennethjiang