kennethjiang/OctoPrint-Slicer #73: OctoPrint doesn't offer to update slicer to v 1.1.1

John-Mc, 2017-03-26:
The latest release of the OctoPrint-Slicer (v1.1.1) is not popping up the dialog box when I log in saying there are updates available. In the past, your updates have always shown up when I first logged in.

If I go to the Software Update section in OctoPrint, It shows that I have 1.1.0 and says "everything is up to date.

kennethjiang, 2017-03-26:
@John-Mc this is a known issue of OctoPrint. From the anecdotal information I heard so far there are at least 20% of the chance when OctoPrint won't prompt for a plugin update.

A workaround is to click *Advanced options* on *Software Update* settings, and click *Force check for update*. Please let me know if OctoPrint still won't prompt for new version after "force check"

John-Mc, 2017-03-26:
_Force check for update_ worked. I had not run into that problem before. In fact I had not even noticed the _advanced options_ choice (but then I haven't been using OctoPrint all that much). Good to know.

closed 2017-03-26 by John-Mc