kennethjiang/OctoPrint-Slicer #74: model move

breadfan13, 2017-03-27:
no option to move multiple stl on bed or auto sort models on bed
[image]

kennethjiang, 2017-03-27:
This is definitely a bug! Does it happen every time you try to use the slicer plugin, or just occasionally? @breadfan13 

Also in case this problem happens again, please try to click on the "rotate" icon and see if you can move the model again.

breadfan13, 2017-03-27:
I can rotate perfectly. And yes, every time it is loaded. Is there a button missing under the three dots? Clicking the empty space there doesn't do anything.

kennethjiang, 2017-03-27:
Can you do me a favor to find out these info:

- Slicer plugin version (in OctoPrint: "Settings" -> "Software Update" -> look for "installed version" under "Slicer".
- Your browser and its version.
- Your operating system and its version

To your question: no there are no buttons missing from your screenshot. the recent version of slicer plugin allows you to just drag the model across the print bed and I'd bet that you will like it much better (if we can figure out what goes wrong) than the older versions.

kennethjiang, 2017-03-27:
Oh, so that you can find a way to use the slicer plugin, can you switch to a different browser? I have done a lot of testing on latest version of Chrome. So please try it in Chrome if that's not what you are already using.

breadfan13, 2017-03-27:
Ok it was my version of Chrome. Must have been an older one. I went to check the version and it auto updated and now I can freely move models. Thank you so much! FYI it was slicer 1.1.1. and chrome version 56. after updating to 57. it solved the problem.

Thank you for your help!

closed 2017-03-27 by breadfan13

kennethjiang, 2017-03-27:
Glad that it worked out for you.

Although I'm also using Chrome 56 I have had no problems. Therefore I'm afraid other users may have the same problem since it's not clear what was causing the problem to you. Are you using a Window, Mac or Linux? Which version?

breadfan13, 2017-03-27:
The bug was on my Windows 10 machine. 

Sent via smoke signal.

John-Mc, 2017-03-27:
When I first upgraded the plugin, I could not move items. I assumed it was user error, and did not have a lot of time to mess around with it. I tried it a couple days later (therefore both my computer and Raspberry Pi had been rebooted) and it was working fine. Unfortunately, I can't remember whether I was running your slicer version 1.1.0 or 1.1.1 at the time. Also, Firefox had an update from 52.0 to 52.0.1 right around this time

I'm running Mac OS 10.12.3, Firefox 52.0.1 (64 bit)

BTW, it really is a great feature - much easier to use than the old method of entering offsets for X & Y.

kennethjiang, 2017-03-27:
Sounds like most people ran into similar issue - it didn't work right after upgrade then it started working after either a force-refresh or system reboot. I will test it out to see if this is the case.