kennethjiang/OctoPrint-Slicer #75: Error Slicing

pook0959, 2017-03-28:
I get an error when trying to slice stl file. where can i check log?

kennethjiang, 2017-03-28:
You can find all the log files in `/home/pi/.octoprint/logs` (assuming you installed using *octopi* image).

However, by default OctoPrint doesn't log Cura commands and debugging info to any of these files. You need to turn debugging on by add `--debug` to appropriate place in `/etc/init.d/octoprint`.

If you don't mind you can send the problematic STL to me and I will debug it for you.

kennethjiang, 2017-03-28:
Also if you get and error message saying that slicing returned with `-11`, it usually means that `cura-engine` core dumped with signal 11.

pook0959, 2017-03-28:
[Responsive_Yo-Yo_(Ball_bearing_axle_and_spacers).zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/877286/Responsive_Yo-Yo_.Ball_bearing_axle_and_spacers.zip)

I was trying to do the left side.

John-Mc, 2017-03-29:
No error slicing that part on my Raspberry Pi running OctoPrint 1.3.2 and slicer plugin v 1.1.1

kennethjiang, 2017-03-29:
These STL work with my setup (the same as @John-Mc's). However I have run into STL files that sliced ok with Raspberry Pi but core-dumped with signal 11 on a ubuntu VM. So I'm not too surprised that @pook0959 has issue slicing the same STL in his OctoPrint.

@pook0959 This is most likely an issue with Cura engine, not Slicer plugin. A few things you can try to work around this error based on my experience: A) upgrade OctoPrint to latest version (which has newer version of Cura); B) rotate and/move the model and slice again; C) add another STL and slice them together. 

Hope one of these options will work for you.

pook0959, 2017-03-30:
Ok, so I had the previous version installed of slicer, and did the upgrade and I realized that is when i started having issues. so I uninstalled slicer, rebooted octoprint and reinstalled the latest version and haven't had issues since. Thanks for the suggestions.

closed 2017-03-30 by pook0959