kennethjiang/OctoPrint-Slicer #76: Dimensions & Lay Object Flat Function

GfiveO, 2017-03-30:
Awesome plugin! I teach 3D Design and Printing to middle school kids. With the limited resources we have and the complexity of the software stack to go from design to printing. Its so much easier for the students to load there models straight to the Octoprint server for each printer. We have four different types. I can setup the slicing profile for each printer and they can just upload multiple projects, position and slice. Awesome. 
With that said, It would be nice to have the dimensions of the object while scaling. Also do I like that when you rotate the object it tries to lay flat, But it would be nice to have a button.

Thanks

kennethjiang, 2017-03-30:
Good timing! I just started to implement the "lay flat" feature, which is quite difficult and would probably take a while before I can get it to work. Showing the dimensions should be easier. If you are happy with just showing L/W/H next to the % when *"scale"* button is activated, it'll probably just take a few hours when I get around to that. Stay tuned!

John-Mc, 2017-03-30:
I agree that showing dimensions when scaling would be helpful. Generally, the kids I work with do all of their scaling in their design program, but sometimes there is an "oops".

Lay flat is a biggie if the kids are using TinkerCad, since it has no such command, nor (unless I missed it) does it even have a way to measure the angle your base makes with the print bed so you can manually lay it flat (it does have a "drop" command which will move the selected object up or down till the lowest point just contacts the bed). 

I had no idea "lay flat" was so complicated. I can see the concept can be tough: figuring out which face is the bottom, then rotating and possibly dropping/lifting to match the bed plane. I just assumed since it was a function in Cura's desktop version, it was a matter of figuring out how to send the command to the Cura plug-in. (Good thing I'm not the one trying to program this sort of thing.)

John-Mc, 2017-03-30:
Kenneth - I just want to thank you for all of your efforts on this plug-in. It has been a huge boon to the classroom where I volunteer. Since they are using TinkerCad on ChromeBooks it has greatly simplified their work flow. They can now do 95+% of their slicing right on OctoPi running on a Raspberry Pi 3 and accessed directly from their Chromebook. We have three basic slicing profiles for each filament. With your plugin, they can tweak those if needed and go. Prior to that, if they wanted anything other than "plain vanilla" slicing settings, they had to export their design from the Chromebook to an ancient laptop running Cura, slice, then bring that file over to the printer (the alternative of having a myriad of slicing profiles for each filament to choose from in OctoPrint was not a great one.)

closed 2017-04-06 by kennethjiang

kennethjiang, 2017-04-06:
@GfiveO @John-Mc I just released 1.2.1, which includes features requested in this issue and better (I hope) user experience.

Lay flat may run pretty slow for models > 5MB, especially if you are on Firefox or IE. In any case please open an issue if/when you run into problems.

John-Mc, 2017-04-06:
Never realized that Firefox was any slower than the others. Is Apple's Safari browser any better? You just might push me to Chrome yet.

kennethjiang, 2017-04-07:
Some how Safari is slightly more sluggish when a huge (in terms of file
size) model across the printer bed. But for the math to "lay flat" model,
Safari comes up slightly ahead, followed by Chrome, and Firefox is
significantly slower. I guess each browser has its own sweat spots and
weaknesses

GfiveO, 2017-04-28:
Hi Kenneth

Thank you for adding the lay flat and dimension to your slicer! My classes have been using it for the last two weeks and our print fails are down. 
I see you have added fan control now too! 

Thanks again
Aaron

FabLab Hawaii
Mary, Star of the Sea School
808 478 2588
fablabhawaii@gmail.com