kennethjiang/OctoPrint-Slicer #77: Filament Flow Setting without effect

HobbyIngenieur, 2017-04-02:
Hi,

after changing filament to a different type, I had the need to increase the Filament Flow setting.
First I tuned it from 100 to 110% and printed unattended, then I set it to 120% because the 110% result was still not okay. During printing the 120% version, I watched my printer and the display stated "100% flow" all the time while printing. The result looks identical to the 100% and 110% prints.

I'm using OctoPrint 1.3.2 with your Slicer 1.1.1 to control an Anet A8 (without any software modifications) and I would be happy, if you could look into this :)

Kindest regards
Jo

kennethjiang, 2017-04-02:
Sounds like a pretty serious bug.

Are you using Cura or Slic3r?

HobbyIngenieur, 2017-04-04:
I'm using Cura 15.04.4 on my Desktop to generate presets and the cura/octoprint integration on the raspi.

HobbyIngenieur, 2017-04-04:
I tried some more ways to change Filament Flow and after these tests, I'm not finally sure, if there is a bug in your slicer but in the cura implementation. What did I do:

1. With a cura generated printer profile (100% flow) I tried to override this setting with setting Filament Flow to 110% in your plugin. Result: 100% flow while printing. With that result I opened this issue.

today:
2. With a cura generated printer profile (110% flow) I tried to slice my model in Octoprint. The value of 110% is displayed correctly as Filament Flow in your Plugin. Result: 100% flow while printing.

3. With the cura 110% flow profile I sliced my model on my PC (Cura 15.04.4) and dropped the gcode to octoprint. Result: 100% flow while printing.

Looking into the gcode-Files, I wasn't able to find a definition for the flowrate (imho M221 S110 for 110% flow) in any of it.

As I'm quite new to 3D printing, I'm not sure at whom I should point now ;-)
 
One fix could be to transfer the flowrate command to "Start GCode" section with a matching reset to 100% in the "End GCode" section, as long as one can't be sure, that the profile setting is applied to the printer correctly.

Hope this helps a bit.

Kind regards
Jo

John-Mc, 2017-04-04:
Is there any chance that it is actually printing at 110%, but your printer doesn't "know" it's 110% because the adjustment was made in the amounts extruded before it gets to the printer? That is, the printer doesn't know that it's already been bumped up, since it's not the printer which is making the extra 10% adjustment.

I'm guessing this is not the case, since you mentioned the changes had no noticeable effect, and I would think a 10% change would be noticeable. 

As a test, what happens if you set it at 50%? Is the print affected at all, and what does the printer say the flow rate is?

thjansen72, 2017-04-04:
Wich firmware shows the flowrate? Mine is telling just the  printing-speed...
Well. If you need 5m filament for the print at 100% you'd need 5.5m with 110% - right?
Just check the gcode if it's changing or not...
(G1 Exxx.x must be bigger)

HobbyIngenieur, 2017-04-04:
@thjansen72 The Firmware Version is "FIRMWARE_NAME:ANET_A8_20160525 VERSION:ANET V1.0".
Meanwhile I was able to produce a satisfying result. And as I stated before, I'm kind of a noob :)

@thjansen72, @John-Mc 
I printed my file again, setting the flowrate directly on the printer control panel to 120%. This time, it was somehow okay. Meanwhile I received your comment an looked into the GCode files. And the Extruder Values G1 ... Exxx.x are the exact multiplication results of my factor and the original value.

My steps to increase the flowrate were simply too small for a good result. Printing my 120% GCode with 120% on the printer control panel sums up to 144% flowrate.

So: 
a) the things work correctly and 
b) this issue can be closed.
The parameters are just not visible in a way I expected them to be. So this can be categorized as a Layer-8 or UX problem :)

Thank you for your time and input and sorry for the confusion.
Kind regards
Jo

John-Mc, 2017-04-04:
It does seem odd that you need a 140% flowrate to make it work. If every print is like this, you may want to look at "calibrating your e-steps", but that's a discussion for another place. Glad you got it figured out.

kennethjiang, 2017-04-04:
Thanks @John-Mc @thjansen72 for chiming in and helping me out here! And I'm glad that you folks got this figured out!

closed 2017-04-04 by kennethjiang