kennethjiang/OctoPrint-Slicer #79: Copy/multiply objects

fromagek, 2017-04-06:
In Repetier-Host it is possible to copy/multiply objects. This is a useful feature if you want to print the same part multiple times.

John-Mc, 2017-04-06:
If you install the "Full-Featured Slicer" plug in, it adds a tab called "slicer" that shows a depiction of the print bed. You can load multiple items on the bed at once (or the same item multiple times), and rearrange them by clicking and dragging. It works well.

The plugin also allows you to change a number of slicing options, so you don;t have to import a new profile for every little change you may want to make.

John-Mc, 2017-04-08:
Just realized: you can load multiple instances on the bed **IF** you have the STL file: Just keep clicking on the "slice" button for that STL and it will keep adding the item to the bed. I don't think it's possible if you are working with a g-code file. Since the file is already sliced, it doesn't get added to the bed before printing. (and the g-code already has the information that positions the file on the print bed, so you'd be loading one on top of the other anyway.)

So if you are slicing on another computer and sending G-code to OctoPrint, you should do all of your duplicating there, before you slice it.

kennethjiang, 2017-04-09:
@fromagek Do you want to duplicate STL objects, or objects with only g-code?

fromagek, 2017-04-10:
I would like to duplicate STL files.
If you want to print a small part many times (e.g. 50 times) you have to add them one by one and orient them. This is a bit tedious, however it already works.
Best florian

kennethjiang, 2017-04-10:
Ah I understand your use case now @fromagek . I think I did run into similar situation once of twice. This is easy to implement - will get to it when I have a chance.

kennethjiang, 2017-04-26:
@fromagek I finally got around to implementing this feature. It's now live in version 1.2.5. Please upgrade

closed 2017-04-26 by kennethjiang

fromagek, 2017-05-08:
Works perfectly!
Thx!!!

commit 175c578d (2017-04-26): Closes #79. Duplicate model