kennethjiang/OctoPrint-Slicer #80: "Split into parts" hangs

John-Mc, 2017-04-06:
I tried to "split into parts" an object which had had had problems with the internal components getting "blown out" under a previous version of the "Split" command. The slicer tab is showing the bed and all selections grayed-out and a graphic with 9 little boxes which appear and disappear over the center of the object.  I assume that means it's calculating. Unfortunately, it's been in that state for about 35 or 40 minutes now. The STL was 2.3MB (a big file for me, but not as huge as what some people deal with).  The previous command which blew out the internal parts only took a couple of seconds on this.

Not sure what's going on, or even the best way to abort this... I'd rather not trash my installation.

John-Mc, 2017-04-06:
Quite my web browser and restarted. Never did anything to the Pi. The part is is no longer on the bed, and things seem to be functioning normally.

Further info: this was running on a "vanilla" installation of Raspian to which OctoPrint was added, not on an installation which came from OctoPi.

kennethjiang, 2017-04-06:
Thank you @John-Mc for reporting this bug. It apparently impacts all users. Am working on a fix and it'll go out soon with 1.2.1.

John-Mc, 2017-04-06:
Version 1.21 fixed the hang, but it's still blowing out the internal parts. The STL is not one I designed, so I can't tell you much about it, but I don't think it's got any loose parts inside.

kennethjiang, 2017-04-07:
I have yet to figure out a good way to solve issue #67. 

On the other hand, I found that Cura has similar behavior in split model into parts. If you get a chance to try your STL with Cura, let me know if it does the same.

closed 2017-04-07 by kennethjiang

John-Mc, 2017-04-07:
My version of Cura is customized by Lulzbot. I can't find any "split" command. 
Downloade Cura 2.4, but can't find a split command in that

kennethjiang, 2017-04-07:
You are right! I have been using Cura 15.04 and found "split into parts"
handy when I had an STL file that contained several unconnected parts.
Somehow newer version of Cura removed this feature.

It happens to me quite a few times when I downloaded an STL from
thingiverse that has several parts, and I wanted to print just some parts,
or want different parts in different orientation.  How often (or ever) did
you run into similar situation?

John-Mc, 2017-04-07:
The Lulzbot version is based on either 15.03 or 15.04, but I can't find "split" in there either. We don;t run into it much at the school, because the kids tend to design 1 thing at a time. When we were running Cura on the Windows PC (Lulzbot version) I don't think we had access to it - or if we did, we didn't know that we did. 

It would definitely be useful if we did more downloading from Thingiverse or similar sites. We also did have one instance where one of the kids designed a wheel, made 3 more copies in TinkerCad, and exported the whole thing. Unbeknownst to him, he had tilted the first wheel, so all four wheels were tilted. When it started printing, the first layer was only printing abut 1/3 of the wheel (no supports, and only part of each wheel was touching the bed). If we had a split command, we could have split into four wheels, and laid each one flat. Without it, we had to go back to TinkerCad, delete 3 of the wheels, export single wheel & load to Cura, Lay Flat, load another & lay flat, and so on.

So I can see that this would definitely be rather useful for some people. Just with our own typical work flow it doesn't happen much.

If the automatic split is causing fits, maybe a way to manually select an area and say "split this off" would be a stopgap while the automatic sequence is running?

In fact, there may be instances where that would be preferable? I have a ball-inside-a-cube STL that I printed for the kids just as a curiosity and an example of something that you can make with 3D printing, but not with some other means. The ball prints inside the cube, but is completely unattached. "Split" separates the ball and the cube, which is not what I would want to happen if something like this were in an STL with other objects.

John-Mc, 2017-04-09:
I could not find "split into parts" on the Mac OS version of Lulzbot's customized Cura, (based off of 15.02.01). I've downloaded and checked both versions 15.04.6 and 2.4 for Mac OS (regular Ultimaker versions, not customized for Lulzbot printer) - I can't find a "Split" command there either. I thought perhaps it was that Cura would not display the command if there was nothing to split (I had loaded a very simple STL), so I made an STL that was two separate pieces and loaded it. I'm still not seeing "Split" as an option anywhere, so I don't have much to compare it to.

The plug-in's split command did work perfectly on an STL I imported .consisting of two simple objects with no internal parts.

kennethjiang, 2017-04-09:
Can you upload the model with 2 simple parts? I will take a look at why "split" doesn't work on it.

Attached is the screenshot of my Cura version of 15.04.6 (Mac) with "split object into parts" command. Interestingly it also blows out internal parts for the couple of models I just tested. 
[image]

John-Mc, 2017-04-09:
Kenneth - I believe you misread my earlier post. Your slicer's split command worked **perfectly** on my model with 2 simple parts. I'm assuming you don;t actually need that model, given that it worked just fine.

Thanks for the screen shot of Cura Mac. Somehow I had missed that the split command was called up by control-clicking on the object (which I assume is Right-clicking in Windows). I was looking for it in the menus or buttons.  

Now that I know where the command is, splitting with Cura 15.04.6 on the Mac has the same effect on an STL as your plugin's split command does (works well on simpler parts with no complicated internal pieces, but blows out the internals of the more complex assemblies). And it looks as though they've pulled that command from Cura 2.4

kennethjiang, 2017-04-09:
Sorry. I did misread your comments. I wonder why cura pulled this command
from newer version. It seems that there are quite a few users in the same
position as me to need to use 'split' once in a while. When version 1.2.0
broke this function they ran into problems​.