kennethjiang/OctoPrint-Slicer #81: Lay Flat command resets z-axis rotation

John-Mc, 2017-04-08:
I just noticed a minor bug in the Lay Flat command: 

If I rotate an object around the z-axis, then choose "Lay Flat", the object get laid lflat, but it also gets returned to its original z-axis orientation. I think it would be preferable if Lay Flat left the z-Axis rotation alone.

It's a minor bug, since the work-around is obvious: choose Lay Flat first, then do any z-axis rotations after.

kennethjiang, 2017-04-09:
This turns out to be harder to fix than I thought.

The tricky part is that z-rotation will impact x- and y- rotation. To see how it works, if you rotate a model to along x and y to any angles other than multiples of 90 degrees, then when you rotate z, you will see that x and y rotation will change with it.

So it looks to me that fixing this bug is to solve this math problem: given original rotation (X, Y, Z), and the new optimal rotation (X', Y', Z'), find a (X", Y", Z) that is equivalent to (X', Y', Z'). 

Let me know if I over-complicated this problem...

kennethjiang, 2017-04-15:
Sorry for the delay.

You are probably right that lay-flat can be done just by rotating X and Y, however the way that how lay-flat is calculated is by using the "orientation", the vector that is perpendicular to one of the faces on the model. I then used a function in three.js to convert the orientation back to rotation, but that function doesn't seem to have a way to fix Z rotation while trying X and Y. There is probably a way and it's probably not that hard but it's a bit beyond my current geometry knowledge...

John-Mc, 2017-04-15:
Understood. Probably not worth further investigation. As mentioned in first post, we can always just do whatever z-axis rotation is needed after the lay flat

kennethjiang, 2017-04-24:
@eyal0 You seem to have done something interesting around lay-flat/tipping. Should I pull your branch and take a look at it?

eyal0, 2017-04-25:
If you like, sure! I'll tell you what I did:

What I did was calculate the model's center of mass (centroid) and also the convex hull of the points that are touching the platform.  If the centroid is over the convex hull then the object won't tip over.  But if the centroid isn't then I rotate the object until it is touching the platform at a new point.  I repeat this until the centroid is over the base.

This should be similar to how an object would actually fall if you place it on the ground.  It seems to work differently than the current tip.  For example, try this file:

http://www.thingiverse.com/thing:1067068/#files

My algorithm is still slow and not well tested so I didn't offer it yet.  You can look around if you want.

eyal0, 2017-04-25:
Mostly it's really slow so I'm working on that first.

eyal0, 2017-05-02:
@kennethjiang If you want to fix up the z axis issue, maybe you can convert your Euler rotation from "XYZ" order to "ZYX" order first, then lay flat, then convert back to "XYZ".  I did this with model arranging to solve the same problem.  Three js docs explain how to change the order of Euler rotation.  It just does some matrix math so it's a fast calculation.