kennethjiang/OctoPrint-Slicer #84: Regression: lock "scale" doesn't work

kennethjiang, 2017-04-10:


John-Mc, 2017-04-10:
Seems to be working on my installation. If the "lock" check box is selected and I type 1.5 in the x axis scale box, the other two axes change to 1.5 as well. If it's unchecked, changing the scale in one box changes only that axis.

kennethjiang, 2017-04-11:
hmmm this regression was introduced by 1.2.0 or 1.2.1. You must be on 1.2.1 so I'm wondering why you are not affected by this bug...

closed 2017-04-11 by kennethjiang

John-Mc, 2017-04-11:
I'm using 1.2.1  Just tried it again to make sure. Working fine.

kennethjiang, 2017-04-11:
Aha I tried it on Firefox and I didn't have the bug. So it's on Chrome and Safari only! The browsers still have not-so-subtle differences.

kennethjiang, 2017-04-11:
I just released 1.2.2 which fixed it anyway.

commit 3a4d7722 (2017-04-11): Closes #84