kennethjiang/OctoPrint-Slicer #85: Missing table-volume

thjansen72, 2017-04-11:
Hello!
I rebuild my printer for a longer (300mm) y-axis. After changing the parameters and restarting octopi, i realized, the slicer is missing 50mm in back...
I didn't tried it out yet,  to see if it realy makes an issue on the slicers work - but you'll never know...
[image]

kennethjiang, 2017-04-11:
Yes I do see the problem. It's a terrible bug. Will work on a fix ASAP.

eyal0, 2017-04-15:
Might be a duplicate of https://github.com/kennethjiang/OctoPrint-Slicer/issues/86

thjansen72, 2017-04-16:
Jepp. It looks good with the new update.
I also think its a little slower. The STL from the image above (7.7MByte) took 448 sec to slice.. (chrome) - but it's just my view and i don't rembember the exact time of the earlier version of the slicer.
No matter about this. Thanks!

kennethjiang, 2017-04-17:
I just tested a stl file of 10MB and it took only 29s. So 448s is definitely too long.

Are you using Cura or Slic3r? I have noticed that Slic3r takes a lot longer than Cura for the same STL.

If you are using Sli3r, can you attach your STL file so that I can investigate why it takes so long?

eyal0, 2017-04-17:
It's probably just the slicer and not this plugin.

Once the file starts uploading, the plugin is done and its just a matter of
upload speed and slicer speed.

If you open the network tab in the the developer console by pressing F12…
you can see when the uploading starts.

thjansen72, 2017-04-17:
Don't forget I use a bigger bed size! n a 200x200mm sized bed the slicing is verry faster!
[Almuth2.stl.txt](https://github.com/kennethjiang/OctoPrint-Slicer/files/925188/Almuth2.stl.txt)

kennethjiang, 2017-04-17:
@thjansen72 With your STL I had similar results as yours: on my OctoPrint (Raspberry Pi 3) it took almost 300s to slice, whereas on my MacBook Pro (using Cura) it took less than 10s.

The reason why there is such a huge difference is that the Slicer Plugin sends STL file to Cura on OctoPrint server (running on a Raspberry Pi) to do the slicing work. Since Raspberry Pi has much less computing power than desktops, it'll take much longer to slice an STL than Cura on a desktop computer. 

It's good to know that this is not a huge issue for you, since the speed bottleneck is outside of Slicer Plugin and I can't think of any way to improve the situation. But let me know if you have some ideas.

eyal0, 2017-04-17:
I've been thinking of making a client-server slicer so that RPi can send
slicing jobs to other, faster computers, to slice. But then you'd have to
set up a computer that is always on for that. The advantage, I thought,
would be using slicers that don't run on RPi like simplify3D.

5 minutes to slice a file that takes 5 hours to print anyway doesn't seem
like a big deal.

thjansen72, 2017-04-17:
No, it's absolutly no problem on my side. My RPi3 got some other things to do (X-Server with touchUI and 7" touchscreen, VNC, a.s.o.) so I got less  power for calculation. It's working perfet for normal objects and tthat all that matters!
@eyal0  Nice idea. Maybe transfering data via http (post/get) into a little java applet, witch will start the sclicer as "system-command". This could be done on every computer, where a supportet slicer is installed. Seems to be "just" a question of possibele automatic starting the sclicer with all needed arguments...

kennethjiang, 2017-04-24:
@thjansen72 @eyal0 I have thought about building a "slicer cloud" that will do exactly what you talked about. But it is by itself a much bigger scope than the whole Slicer Plugin.

I will close this issue for now. But will keep you posted if/when I'm ready to start on the "slicer cloud".

closed 2017-04-24 by kennethjiang