kennethjiang/OctoPrint-Slicer #88: Non-docker build instructions needed, how to minify slicer.js?

eyal0, 2017-04-15:
How do I build and install without using Docker?  I usually run and test my plugins without docker like this:

./venv/bin/pip install ../OctoPrint-Slicer --upgrade && ./venv/bin/octoprint --verbose

That installs the current code and then runs Octoprint.  It used to work but the steps that copies slicer.js to octoprint_slicer.min.js is missing.  What is the step to minify and concatenate all the input files?

eyal0, 2017-04-15:
I figured it out.  I'm able to reinstall OctoPrint-Slicer and launch OctoPrint with this command:

`cd ../OctoPrint-Slicer && npm run build && cd ../OctoPrint && ./venv/bin/pip install ../OctoPrint-Slicer --upgrade && ./venv/bin/octoprint --verbose`

closed 2017-04-15 by eyal0

kennethjiang, 2017-04-15:
I switched to using Rollup and didn't update README. Sorry!

During development, you may want to run `npm run watch`. It will monitor changes in `src` folder and recompile them to `octoprint_slicer/static/js/octoprint_slicer.min.js` whenever there is change. Makes it convenient. `npm run watch` is not robust. It will crash often. So you will have to relaunch it from time to time.