kennethjiang/OctoPrint-Slicer #91: Print bed grid missing

John-Mc, 2017-04-16:
Just upgraded to 1.2.3.  The grid on the print bed is missing when viewed in Firefox on a Mac. Switched to Safari, and the grid is showing there, but the labels "Front", "left", etc. are showing up twice. Looks normal in Google Chrome (Mac) browser.

Not sure if this is related to the update or not, but OctoPrint seems to be taking much longer to load, and slicing seems to be much slower as well (at least in Firefox). This could be completely in my mind, however.

All conditions noted above have survived shutdown, remove power, and reboot.

kennethjiang, 2017-04-16:
Thank you so much @John-Mc  for catching this! Will push a fix soon.

What version of Safari are you using? I'm using Safari 10.1 on Mac and not seeing the problem of labels showing twice.

John-Mc, 2017-04-16:
Just fired things up again, and the double labels in Safari are gone (Safari 10.1, Mac OS 10.12.4). Not sure what's happening here, since I did reboot before.

kennethjiang, 2017-04-17:
Just pushed 1.2.4. The print bed should be fixed now. Let me know if you still see issues.

closed 2017-04-17 by kennethjiang

John-Mc, 2017-04-17:
Working fine in Firefox now