kennethjiang/OctoPrint-Slicer #99: Minimal layer time

breadfan13, 2017-04-26:
Just a feature request. getting heavy use out of this nifty onboard slicer for my pi3. Could really use a minimal layer time option in settings.

John-Mc, 2017-04-26:
Is minimal layer time something you need to change frequently? I never change mine, so I just set it once in the profiles I export from Cura Desktop, and they are there (I believe) for anything I slice in OctoPrint. I'm fairly sure that just the fact that I can't adjust it in OctoPrint or the slicer plugin doesn't mean that the imported profile setting is ignored.

If this is something you need to change frequently, I'm not objecting to it. We're usually printing fairly basic things in an elementary school classroom. Your needs may be much more complex than ours.

kennethjiang, 2017-04-26:
I actually ran into this situation quite a few times when I do want to tune minimal layer time. Ironically most of the times it was because Cura failed to respect the set value of minimal layer time.

I never quite figured out if my struggles with minimal layer time in Cura were caused by user error, or bugs in Cura. My general experience is minimal layer time seldom works in Cura 15.04, but works 50% of the times in Cura 2.1.3.

@breadfan13 does minimal layer time work for you in Cura 15.04 (the desktop app, not the one embedded in OctoPrint)?

kennethjiang, 2017-05-03:
@breadfan13 not sure if you saw my previous comments - unless there is a way to confirm that minimal layer time actually works in Cura 15.04, there is no point of adding it to the options.

kennethjiang, 2017-05-31:
@breadfan13 I am closing this issue assuming it's no longer a problem for you. Feel free to re-open it if otherwise.

closed 2017-05-31 by kennethjiang