kennethjiang/OctoPrint-Slicer #100: Get rid of modal to confirm model loading

kennethjiang, 2017-04-28:
Instead:
- Add a button to "clear bed"
- Add models straight to the bed with prompting

John-Mc, 2017-04-28:
Not sure I understand this one. You already have a "clear bed" button.
How is adding models straight to bed with prompting different than what is in the release version now? Are you prompting for "add multiple models" and if selected it stays in that mode (allowing models to be added with just a wand click) until you click something to cancel the bulk add mode?

kennethjiang, 2017-04-28:
It was a typo - should read "...**without** prompting".

Also I updated the wording about the button. The intention is to move current "clear bed" button to top level, probably right below current "remove" button.

Appreciated your questions @John-Mc  and my apology for the sloppiness.

eyal0, 2017-04-29:
This is what existed months ago. Add was always without clearing the bed. The only problem that my users had with that is that sometimes they wouldn't notice that they are printing multiple files.

kennethjiang, 2017-04-29:
It is very similar to what I had before added the modal. The only difference is now there will be a "clear bed" button, hopefully very discoverable.

Can you elaborate what would cause your users to print multiple files without noticing it? Is it because they load a new STL file and move right into pressing the *Slice it!* button, not noticing that the bed is not empty to begin with?

eyal0, 2017-04-29:
Yes, that's what happened.  But that was before the auto arrange and he had a small STL overlapping a big one so it was less noticeable.  Now that there is auto-arrange, it's easier to see when there are two models.

commit 1831090f (2017-05-17): Closes #100. bump version
closed 2017-05-17 by kennethjiang