kennethjiang/OctoPrint-Slicer #104: Split into parts cannot be sliced

eyal0, 2017-04-30:
To reproduce, load a model that can be split.  For example: http://www.thingiverse.com/thing:2020668/#files

Then split parts (in the ... menu) and then try to slice.  The slice option will be unavailable.

Broken on version 1.2.5.

eyal0, 2017-04-30:
Works on [1.2.4](https://github.com/kennethjiang/OctoPrint-Slicer/releases/tag/1.2.4). So it must have broken in between the two.

eyal0, 2017-04-30:
Using git bisect between 1.2.4 and 1.2.5, the bad commit is found:

```
175c578d161f5247cc30ec9718e397c24ef5f50d is the first bad commit
commit 175c578d161f5247cc30ec9718e397c24ef5f50d
Author: Kenneth Jiang <kenneth.jiang@gmail.com>
Date:   Wed Apr 26 09:08:07 2017 -0700

    Closes #79. Duplicate model

:040000 040000 2284cfb60ba73214cbdf681ea9714789d321dc7b e5c6cde0207f5a7a15d5c61a38b49cadeb10d4e8 M      octoprint_slicer
:040000 040000 eeb8d29c4874a3030cda59a2b41c88c437c0dd54 e1839fbb3ba9bd1155e4051961640a417cd7ac30 M      src
```
https://github.com/kennethjiang/OctoPrint-Slicer/commit/175c578d161f5247cc30ec9718e397c24ef5f50d

eyal0, 2017-04-30:
The problem is due to deleting the model before adding the new ones.  Because of this, resetSlicingViewModel is called for the delete but setSlicingViewModel doesn't get called.

setSlicingViewModel expects to get a target (local or sd) and a file name.  We don't have those when a model is split because only the original model has a file name.  The new models after split don't.

I suggest returning to the previous behavior of first adding and then deleting.  This is an easy solution.

kennethjiang, 2017-04-30:
Thanks a lot @eyal0 for not only report this bug but also pointing out the root cause. Greatly helped me with the fix!

closed 2017-04-30 by kennethjiang

eyal0, 2017-04-30:
Oops, I made a fix at the same time.  With your fix, you are arranging models after delete, which I think would be unexpected for the user.  Also, with the fix that I proposed, I am not calling model arrange after each add in the split so performance is improved.

Perhaps consider using my fix instead...

kennethjiang, 2017-04-30:
If models are not re-arranged after delete, there will be a "hole" after a model is split, because it would go like this:
1. model A is split into 5 parts
2. 6 models in total on the build plate
3. 6 models are re-arranged
4. original model is removed without re-arrange
5. there will be a "hole" at where the original model is

As you probably have found out, this part of the code (adding/removing models, making sure `slicingViewModel` is always in valid state, making sure g-code file name is handled correctly) is already complicated and bug-prone. Any slight change is likely to result in bugs. That's why I have been trying very hard to keep things (including code) simple. Like I commented in your PR, your fix saves a few method calls but it's not worth the increased complexity and risk of bugs.

commit 3d656c8d (2017-04-30): Closes #104. Remove model needed to be called after split parts are added. Otherwise  will be missing for