kennethjiang/OctoPrint-Slicer #107: Get rid of analytic javascript libraries and tracking of users

thantik, 2017-05-01:
Currently the plugin reaches out to the internet in order to download 2 js analytic libraries.  I'm not sure what you're tracking, or why you think it's appropriate to track users of the plugin, but it needs to be removed.  We already fight for our privacy a million other places, and it's not appropriate to be tracking OctoPrint users via a phone-home analytic library like this.

As it currently stands, if the printer is not connected to the internet, the plugin breaks and falls back to the non-gui slicer.  So it also happens to break functionality.

kennethjiang, 2017-05-01:
The js libs help me understand where users get stuck, and what bugs my code has so that I can fix them. It is one of the best practices in writing software. Without these essential tools this plugin would have a lot more bugs then it does today.

If your concern is on your privacy - rest assured there is no way for this plugin to track your personal info. There are built-in mechanism in HTTP/HTML protocols that prevent any web page/javascript from doing such thing. Google or Facebook may have a lot of your personal info but there is no way for these 2 js libs to get access to them.

If your concern is that these 2 js libs would prevent this plugin from working correctly when it's offline, the latest version (1.2.5) should have already fixed that. Please upgrade to that version and let me know if it's still broken.

thantik, 2017-05-01:
It looks like we'll be keeping a fork of this project with the analytic libraries removed on our end.  git merge seems to do a pretty good job of ensuring an updated fork with the required libraries being removed from use.  We are not comfortable with our users (many of them schools) being subjected to any kind of analytics.

kennethjiang, 2017-05-01:
@thantik Maintaining your own fork is a good idea if you still have concerns. One of the biggest benefits of open-source softwares. :)

Also feel free to take a deep look at these 2 libs: MixPanel and Raven. They are reputable tools used by other companies including security-sensitive ones like DocuSign and banks.

closed 2017-05-01 by kennethjiang

twilde, 2017-08-30:
Kenneth, would you consider providing an opt-out setting for the use of Raven?

kennethjiang, 2017-08-31:
I'd hesitate to do that - I have tried really hard to keep this plugin as simple as possible. It doesn't even have a settings page. And I have actually removed a few features because they were not easy enough to understand/use for most users. Because of that, I feel that adding an option to opt out Raven will be a big blow on the simplicity of this plugin.

Since Raven doesn't collect any Personally Identifiable Information, I don't feel it's a liability to leave it in. Of course, I understand some users will still have concerns. In that case, do you think it's a viable solution for you to just fork the code and remove the Raven part yourself?