kennethjiang/OctoPrint-Slicer #108: Brim's not being created when selected

Geoff-Ilmax, 2017-05-01:
When i select a Brim to be added to my print, and then look at the gCode or try to print, no Brim is shown or printed.

kennethjiang, 2017-05-01:
Did you select Cura or Slic3r? Can you give me more info about the STL file you tried to slice and the gcode generated?

Geoff-Ilmax, 2017-05-01:
This is the lastest STL file that I was slicing that I noticed it with.
http://www.thingiverse.com/thing:1672959/
I believe it was with Cura as I don't think i have Slic3r installed, but I wll double check once my current print finishes.  Once it completes, I will upload my Slicing profile, and the generated gcode.

Geoff-Ilmax, 2017-05-01:
[image]
[image]
[image]

kennethjiang, 2017-05-01:
I tried that STL with `brim` option and it worked for me (plugin version 1.2.5). Please see the attached screenshots for gcode layer 1 and layer 2.

[image]
[image]

kennethjiang, 2017-05-01:
yes please upload your slicing profile and gcode if you still experience the problem.

Geoff-Ilmax, 2017-05-01:
Here is my profile.  In my profile I have Brim turned off, But I am selecting it when I use the slicer, but not getting it in the gcode that it creates.
[A8-L2-F15-S0-P0.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/968802/A8-L2-F15-S0-P0.zip)

kennethjiang, 2017-05-02:
I tried it with your profile, and then set adhesion to **brim**. The resulted gcode still has brim. I attached the resulted gcode. Please compare it with yours.

[Refuerzo_en_T.gco.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/970711/Refuerzo_en_T.gco.zip)

kennethjiang, 2017-05-03:
I'm closing this issue assuming that you no longer experience the problem @Geoff-Ilmax . You can reopen it if you still do.

closed 2017-05-03 by kennethjiang