kennethjiang/OctoPrint-Slicer #114: Settings in Slicer not saved ?

OctoMoe, 2017-05-05:
I like your Slicer a lot. This is something that was missing in Octoprint for sure.
It happend now twice to me, that i did set infill to 30% in your slicer, then hit the slice button and started to print. The prints came out with 0 infill. Did i go wrong somehow since it does not recognize the 30% ?
When i change the same profile with the same stl  on my phone with a apk i have and slice it there, infill is fine.

kennethjiang, 2017-05-06:
My slicing profile is 20% infill by default but I sometimes (just did it yesterday) set it to 30% and have not noticed any problem.

Can you slice it again with 30% to see how it comes out? If it still doesn't work, can you attach your STL, profile, and generated gcode so that I can look deeper?

Smombie-Rom, 2017-05-07:
What i found out now : 
The standard profile i used was 0 infill. Changing infill to 30% did not save the new settings for me. 
I made a new profile where 30% is pre set, and there it is fine. 
Just sliced again. 
Steps i took, so we can iron out errors:
Loaded the stl file, changed infill from 0 (blank input selector) to 30 and hit "slice it"

I made you 2 gco files. One where i set up 30% by myself, and one where 30% where pre set in the profile. Stl also included in the zip.

Thanks for checking.

[slicing_problem.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/981449/slicing_problem.zip)

kennethjiang, 2017-05-08:
`sliced_with_set_to_30.gco` is definitely corrupted in a weird way. I tried it with your STL and the generated gcode was fine. However I'm using Cura 14.01 in my OctoPrint.

I'm upgrading to Cura 15.04 to test it again. It will take a while. Will keep you posted.

kennethjiang, 2017-05-10:
@OctoMoe I finally upgraded Cura to 15.06. Gcode still came out ok. Attached is the sliced gcode with infill manually set to 30% and the `.ini` file with infill being 0%.

Here is what I suggest you to do: 
1. Uninstall and reinstall Slicer plugin to see if the problem goes away.
1. If it doesn't solve the problem, rebuild the your Raspberry Pi from the latest Octoprint image. You will loose all your data on your SD card unfortunately. I know it's painful but I'm 99% sure it'll solve your problem.

[Archive.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/988401/Archive.zip)

closed 2017-05-17 by kennethjiang