kennethjiang/OctoPrint-Slicer #116: Exception: Cannot read property 'point' of undefined

kennethjiang, 2017-05-15:
Stack trace:
/static/webassets/packed_plugins.js in QuickHull.computeExtremes at line 12890:34
/static/webassets/packed_plugins.js in QuickHull.computeInitialHull at line 12948:33
/static/webassets/packed_plugins.js in QuickHull.compute at line 13306:14
/static/webassets/packed_plugins.js in QuickHull.setFromPoints at line 12641:14
/static/webassets/packed_plugins.js in new ConvexBufferGeometry at line 13684:37
/static/webassets/packed_plugins.js in new ConvexGeometry at line 13662:29
/static/webassets/packed_plugins.js in hullPointsFromGeometry at line 31507:26
/static/webassets/packed_plugins.js in Object.box3FromObject at line 31589:46
/static/webassets/packed_plugins.js in Object.box3FromObject at line 31638:48
/static/webassets/packed_plugins.js in STLViewPort.self.addModelOfGeometry at line 31844:37
/static/webassets/packed_plugins.js at line 31824:18
/static/webassets/packed_plugins.js at line 11765:13
/static/webassets/packed_plugins.js in XMLHttpRequest.<anonymous> at line 8981:298

kennethjiang, 2017-05-15:
@eyal0 This exception happened to one of the users. Do you have any clue what could have been the cause of it?

eyal0, 2017-05-15:
Deep inside QuickHull, which isn't even code that I wrote.  I got it from the three js examples.

Maybe the STL file was invalid?  If it's repeatable, maybe the user can send you the STL file?

Compute initial hull tries to find the initial 4 sided shape to grow into a convex hull. Maybe there are less than 4 points in the shape, which is an illegal shape.  Or maybe the shape is entirely flat, also illegal.

See if you can get the STL file.

kennethjiang, 2017-05-15:
Got it! thx @eyal0 ! This error doesn't seem to happen often. Will keep an eye on it.

kennethjiang, 2017-05-31:
Has not happened frequently. Closing it baed on the assumption it is one-off/browser-specific issue.

kennethjiang, 2017-06-01:
This error occurs at the rate about 1 user/week. Impact all browsers.