kennethjiang/OctoPrint-Slicer #118: Use right-button drag to orbit the view

kennethjiang, 2017-05-22:
Currently left-button drag is used to both move objects and obit the view, just depending on where the mouse button is pressed. This is prone to error especially when users are in a rush to position object and slice it.

Since orbiting the view is used less often than moving objects, the user experience will be improved by assigning secondary-button-drag (right-button drag) to it.

John-Mc, 2017-05-22:
Since Macs have no right button, will you be implementing a keyboard equivalent? I suggest control-click-drag, since the control click seems to be an "almost standard" substitute for a right click on the Mac

kennethjiang, 2017-05-22:
Mac has "secondary click" that's equivalent to right button on windows. See attached screenshot. 

I think Cura and a couple 3D design tools do the same thing: left-drag to move, right(or secondary)-drag to orbit. Personally when I'm on Mac and confronted with similar situation I'd try these in order: secondary-drag, cmd+drag, ctrl+drag, shift+drag, opt+drag. So secondary-drag would have better discoverability than ctrl+drag for users like me. :)

[image]

kennethjiang, 2017-05-22:
Although I'm personally more accustomed to secondary-click/drag than using control, especially on a web page, but I could be in the "minority group". Let me know if you think most users on Mac will find ctrl+drag more intuitive @John-Mc

John-Mc, 2017-05-22:
I had forgotten all about "secondary click". In fact, I've probably missed out on using my trackpad to control some things. My use of Macs long predates the appearance of track pads, so I'm used to some sort of keyboard alternative (fortunately for me, many programs seem to allow keyboard alternatives). If it's covered one way or another, it's fine with me. No need to make accommodations for my dinosaur methodology.

kennethjiang, 2017-05-22:
Hehe blame Steve Jobs for his hate for the right button. Although his engineers managed to sneak the function in it was never promoted. ;)

commit dbac10c6 (2017-05-31): closes #118. Use secondary drag for orbiting
closed 2017-05-31 by kennethjiang