kennethjiang/OctoPrint-Slicer #119: issue on installation

Tdoc31, 2017-05-24:
Hi

Love your plugin, but cant install it again

in the install window the following

Installing plugin...
Installing plugin from https://github.com/kennethjiang/OctoPrint-Slicer/archive/master.zip...
/home/pi/oprint/bin/python -m pip install https://github.com/kennethjiang/OctoPrint-Slicer/archive/master.zip
Collecting https://github.com/kennethjiang/OctoPrint-Slicer/archive/master.zip
Downloading https://github.com/kennethjiang/OctoPrint-Slicer/archive/master.zip (1.5MB)
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-Slicer==1.2.8 from https://github.com/kennethjiang/OctoPrint-Slicer/archive/master.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied (use --upgrade to upgrade): OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg (from OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): sockjs-tornado<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): requests<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): semantic-version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.1-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): websocket-client<0.41,>=0.40 in /home/pi/oprint/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in /home/pi/oprint/lib/python2.7/site-packages/MarkupSafe-0.23-py2.7-linux-armv7l.egg (from Jinja2<2.9,>=2.8->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): certifi in /home/pi/oprint/lib/python2.7/site-packages/certifi-2016.02.28-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): backports.ssl-match-hostname in /home/pi/oprint/lib/python2.7/site-packages/backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): blinker in /home/pi/oprint/lib/python2.7/site-packages/blinker-1.4-py2.7.egg (from Flask-Principal<0.4,>=0.3.5->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): Babel>=1.0 in /home/pi/oprint/lib/python2.7/site-packages/Babel-2.3.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): speaklater>=1.2 in /home/pi/oprint/lib/python2.7/site-packages/speaklater-1.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): webassets>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/webassets-0.11.1-py2.7.egg (from Flask-Assets<0.11,>=0.10->OctoPrint->OctoPrint-Slicer==1.2.8)
Requirement already satisfied (use --upgrade to upgrade): argh>=0.24.1 in /home/pi/oprint/lib/python2.7/site-packages/argh-0.26.1-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->OctoPrint-Slicer==1.2.8)
[... cut ...]
Requirement already satisfied (use --upgrade to upgrade): pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages/pytz-2016.3-py2.7.egg (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-Slicer==1.2.8)
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Done!

but in the octoprint top right I have the message below in red
Something went wrong
Installing the plugin from URL "unknown" failed: Could not parse output from pip

dont know what to do ??

thanks for any form of help

kennethjiang, 2017-05-24:
From the messages it looks like Slicer plugin was already installed successfully. If you haven't done it, can you reboot the Raspberry Pi and check if Slicer plugin is installed?

Tdoc31, 2017-05-25:
thanks for your reply
several reboot and it is not in the list of installed plugin
Anychance for a command line to have a force uninstall ?

thanks

kennethjiang, 2017-05-25:
If you can log into your Raspberry Pi you can do `sudo /home/pi/oprint/bin/python -m pip uninstall https://github.com/kennethjiang/OctoPrint-Slicer/archive/master.zip` to uninstall the plugin. However, I don't have a high confidence that re-installing it will work. Seems like your OctoPrint is in a little funky state that causes it to ignore plugins that are already installed.

What's your OctoPrint version?

Tdoc31, 2017-05-27:
I will reinstall octoprint as a whole and see

kennethjiang, 2017-05-30:
Let me know if the issue persists after you reinstall OctoPrint.

Tdoc31, 2017-06-01:
Hi

All is fine, even get an update this morning 1.2.9 if I recall well

thanks
Ps this thread is closed

closed 2017-06-01 by Tdoc31