kennethjiang/OctoPrint-Slicer #120: Remove original color in STL files

kennethjiang, 2017-06-01:
If STL files contains color data it'll mess up how models are presented. Color needs to be removed before loading into Slicer.

closed 2017-09-05 by kennethjiang

John-Mc, 2017-09-05:
Does this affect those who are printing with multi-color printers?

kennethjiang, 2017-09-05:
Sorry @John-Mc for the misleading title. The symptom of the issue was that some models were displayed with color/texture different from expected. Originally I thought it was caused by custom colors in STL. It turned out that the issue was caused by incorrect normals in STL. Therefore my fix didn't touch anything about color. Instead I just recalculate normals upon loading an STL.

I changed the title and description of this issue to reflect the real issue and fix.