kennethjiang/OctoPrint-Slicer #121: Work with profiles uploaded from Cura 2

kennethjiang, 2017-06-12:


John-Mc, 2017-06-12:
This would be great, but wouldn't we need a Cura 2.x plugin first?

kennethjiang, 2017-06-12:
Creating a Cura 2 plugin is a much harder endeavor. Also the best place to bring in Cura 2 is OctoPrint core, which apparently has been on Gina's mind for a while now.

However, with more and more Cura 2 users, some Cura 2 profiles have been uploaded to OctoPrint. Since Cura 2 profiles are incompatible with those of older version, Slicer plugin is broken. What I'm trying to address with this issue is to figure out a way to make Slicer plugin play nicely with Cura 2 profiles, but not Cura 2 itself.

kennethjiang, 2017-06-12:
Lower the priority since this doesn't seem to be a use case that regular Cura 2 users will run into. Instead, the problem happens only when users manage to upload an incomplete Cura 15.xx profile to OctoPrint.