kennethjiang/OctoPrint-Slicer #124: Import Stl From local files

tobiaspedersen, 2017-07-04:
I can't find any method of importing a file from locally uploaded files, only when drag and dropp to the web interface. It would be a nice feature to have if it is not already there

thjansen72, 2017-07-04:
You can load an STL in the slicer by hitting on the symbol in the file list.
If you're file isn't listed, it's in the wrong path.
I think the  possibility of loading lokal files, could be implemented in the main Octoprint process (ie by adding the path)  - but is missplaced in the slicer.

kennethjiang, 2017-07-07:
Hi @tobiaspedersen Did @thjansen72 's comment answer your question?

thjansen72, 2017-07-09:
It seems not to be "wanted" by octoprint itself (I think i know why). Even when using  the filemanager-plugin you can only choose from files  uploaded via the web interface first. And you're only able to browse in the local path to the octoprint-user-files.
I think this is made because of the multiple OS octoprint is running on...
So, the best place for your question would be:
https://github.com/foosel/OctoPrint

p.s. have you tried  to create  a link  to your local files in the octoprint folder ....

John-Mc, 2017-07-09:
If I'm following the original request lately, you may want to look in to setting up a "watched" folder. OctoPrint does not like you putting files directly into the uploads folder. It will watch the Watched folder and automatically upload any files that are left in there.  For info on setting up the Watched folder, take a look at video 3.1 in this series: 
[https://www.youtube.com/playlist?list=PLGc0ilxmLZs8UwoQMdpOkYF4gZMMu_Qro](url)

thjansen72, 2017-07-09:
@John-Mc  Upps. I forgot about that Thank you! Maybe because i don't like watched directorys ...

kennethjiang, 2017-07-09:
Thanks @thjansen72 @John-Mc for commenting on this issue. I think now I fully understand the OP's question. Agreed that this is not a function that should be implemented in Slicer. 

I'm personally not a big fan of how file management works in OcotPrint. My ideal is to have all STL/gcode managed in the cloud. Similar to how google drive works (IMO the usability of google drive is so much better than any file system out there). But I understand that not all OctoPrint installations have constant internet connections.

@tobiaspedersen Do you still have comments/questions on this issue?

tobiaspedersen, 2017-07-09:
Sorry for the late reply. Thanks for all response. I think I described my question poorly, sorry about that. 

@thjansen72 answered my question I the first reply. I couldn't figure out how to get a file from the file manager into slicer.

 Thank you

closed 2017-09-03 by kennethjiang